olikraus/u8g2 #1186: Can not user with ESP8266 board (NodeMCU v2)

Kroshtandt, 2020-05-09:
Hello, I've tried to use this library to connect my ST7920 to ESP8266 board NodeMCU but without success.. The screen is blank. I've used my Uno to check if library work - and yes, with arduino controller all works fine. 

What I have done: 
- used both SPI interfaces HW and SW, 
- used logical level converter 
- tried to change spi_mode to 0 or 3 in u8x8_d_st7920.c file.. Nothing works. 

I really out of ides what I can do! I've also both I2C converter (MCP23017) but I haven't found any functional library to work with this display. Just seems that there is no good working library for this display in whole internet.. 

I'm trying to flash simple hello world example

AnHardt, 2020-05-10:
At 5V [ST7920](http://www.hpinfotech.ro/ST7920.pdf) is slow (Page 43 upper table). 
At 3.3V it is even slower (Page 43 lower table).
If the processor is only 3.3V use level converters (and display VDD 5V) or slow down SPI communication.
But even is you are sending a bit to fast you should see, at least, something.
(Provided the wiring is correct at all and the display is not broken.)

Kroshtandt, 2020-05-10:
@AnHardt hello, thanks for your answer. See my wiring:
Using this board:

[image]

and I'm using
SW:
clock: GPIO6 (6)
data: GPIO8 (8)
chip select (CS): GPIO11 (11)
reset: U8X8_PIN_NONE

HW:
clock: GPIO14 (14)
data: GPIO13 (13)
chip select (CS): GPIO15 (15)
reset: U8X8_PIN_NONE

SO if the problem may be with sync of SPI connection, is there any possible fixes for that? Maybe some software settings in SPI, or something like that. I'm already use two kind of converters:

[image]

and TXS0108E module

[image]

olikraus, 2020-05-10:
I also think that a 3.3V to 5V converter is required. Additionally: DId you setup the ST7920 display correctly? You need to ensure, that the display is in SPI mode. Additionally a contrast voltage is required (I do not remember details, but it should be in the datasheet of the display)

Kroshtandt, 2020-05-10:
@olikraus hello, thank you for response! Actually I'm sure that I'm using right pins. 
RS (4) connecting to board pin called MOSI
R/W (5) to CS
E (6) to SLK 

I have connected PSB (15) pin to ground, and setup a right contrast with Uno, so all should be fine with ESP contrast, I suppose. The A and K connected to +5 line and I'm using 200 ohm resistor for A line.

olikraus, 2020-05-10:
The setup should be similar to this picture: https://duino4projects.com/wp-content/uploads/2018/08/Interfacing-Graphical-LCD-ST7920-with-Arduino-schematics.jpg

Kroshtandt, 2020-05-10:
@olikraus sure, what does this port? 
I can try in a few hours, but it work perfectly with Uno without this port.. I hope, that this is will may affect :o

Also I have lolin32 module (ESP WROOM 32), with build-in oled.. I will try to connect screen to it also.. This should be similar, no?

Kroshtandt, 2020-05-10:
BTW, I know that this is weird, but I can try to use, lets say, arduino pro mini as converter between ESP and ST7920. haha. I dont know why I bought this I2C converter, and why they are selling it because it's just useless for this screen..

olikraus, 2020-05-10:
Ok, if the display works with a 5v Arduino, the issue is the level converter as mentioned above.

Kroshtandt, 2020-05-10:
@olikraus @AnHardt don't trust to everything what you read in internet.. The problem was with wrong connection to screen module. The right pins is:

[image]

from datasheet: https://www.digole.com/images/file/Digole_12864_LCD.pdf

Kroshtandt, 2020-05-10:
@olikraus but one more questiomn. Which pins you are using for hardware SPI in case when you are pass only CS pin number? it's a U8G2_ST7920_128X64_F_HW_SPI.

Because screen works only U8G2_ST7920_128X64_F_SW_SPI where I put default hardware SPI pins

Kroshtandt, 2020-05-10:
@olikraus testing with my modules. For you gallery

[image]

[image]

[image]

olikraus, 2020-05-10:
Did you figure out the hardware pins? I actually I do not know the hardware pins. They are defined by your microcontroller.

Kroshtandt, 2020-05-10:
@olikraus HW spi constructor isn't working with ESP32 module for me, but it works with SW constructor where I'm passing HW pins (as I can see in pinout). A little bit weird, but this is not important in my case for now. I wanna use ESP8266, not ESP32, it was connected just for testing purposes. 
ps. both hw and sw constructors work with ESP8266

olikraus, 2020-05-11:
So as a conclusion (and maybe description for the gallery):
- ESP32 Board
- 5V ST7920 directly connected
- U8G2_ST7920_128X64_F_SW_SPI Constructor

right?

Kroshtandt, 2020-05-11:
@olikraus yes, for Wemos Lolin 32 pins are 13 (MOSI), 14 (SCLK), 15 (CS). (only SW constructor for now).

For NodeMCU (V2 in my case) board hardware pins are the same, and both constructors works.

olikraus, 2020-05-13:
Ah, ok, thanks.

closed 2020-12-25 by olikraus

gordon0001, 2022-09-19:
for esp32 support i had to lower the spi frequency in u8x8_d_st7920.c

`static const u8x8_display_info_t u8x8_st7920_128x64_display_info =
{
  /* chip_enable_level = */ 1,
  /* chip_disable_level = */ 0,
  
  /* post_chip_enable_wait_ns = */ 5,
  /* pre_chip_disable_wait_ns = */ 5,
  /* reset_pulse_width_ms = */ 1, 
  /* post_reset_wait_ms = */ 6, 
  /* sda_setup_time_ns = */ 20,		
  /* sck_pulse_width_ns = */  140,	/* datasheet ST7920 */
#if defined(ESP32)
  /* sck_clock_hz = */ 940000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
#else 
    /* ST7920+Due work with 1MHz but not with 2MHz, ST7920+Uno works with 2MHz */
    1000000UL,
#endif
#if defined(ESP8266) || defined(ESP32)
 0,
#else
  /* spi_mode = */ 3,		/* active high, rising edge, 18 Aug 16: changed from 1 to 3 which works for 101  */
#endif
 /* in theory mode 3 should be correct  */
  /* i2c_bus_clock_100kHz = */ 4,
  /* data_setup_time_ns = */30,
  /* write_pulse_width_ns = */ 40,
  /* tile_width = */ 16,
  /* tile_hight = */ 8,
  /* default_x_offset = */ 0,
  /* flipmode_x_offset = */ 0,
  /* pixel_width = */ 128,
  /* pixel_height = */ 64
};`