olikraus/u8g2 #1187: invalid start of buffer (garbage pixel on LCD) ST7565

selevo, 2020-05-10:
It is not erased.
I think the beginning of the buffer is set incorrectly and therefore the cleaning function does not affect this part of the screen.
In which file can I fix this?
(I'm not an experienced programmer)
[image]

[code]
/*

  HelloWorld.ino
  
  "Hello World" version for U8x8 API

  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2016, olikraus@gmail.com
  All rights reserved.

  

*/

#include <Arduino.h>
#include <U8x8lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif

  U8X8_ST7565_ZOLEN_128X64_4W_SW_SPI u8x8(/* clock=*/ 2, /* data=*/ 1, /* cs=*/ U8X8_PIN_NONE, /* dc=*/ 3, /* reset=*/ 4);
// End of constructor list

void setup(void)
{
    pinMode(6, OUTPUT);
  digitalWrite(6, HIGH);
   
  u8x8.begin();
  u8x8.setFlipMode(1);
  u8x8.setPowerSave(0);
  u8x8.setContrast(40);
  
  
}

void loop(void)
{
 
  u8x8.setFont(u8x8_font_chroma48medium8_r);
 u8x8.drawString(0,0,"Hello World!");
  delay(1000);
  u8x8.drawString(0,0,"TEST2");
  delay(1000);
}
[/code]

olikraus, 2020-05-10:
Maybe another ST7565 constructor works better. They do indeed differ in the left position. Please test some of the other constructors. Adjust the contrast if required.

selevo, 2020-05-10:
I checked others, but there the image of the letters is mirrored.
But there is no chaos of pixels.
[image]

U8X8_ST7565_LX12864_4W_SW_SPI

olikraus, 2020-05-10:
There are many more to try. Please test the following devices also:

https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-64128n
and
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7565-nhd_c12864

selevo, 2020-05-11:
But  this graphical...

olikraus, 2020-05-11:
Yes sure, u8x8 will also exist

selevo, 2020-05-11:
I looked, there is nothing suitable.

olikraus, 2020-05-11:
I would expect that the constructor 
U8X8_ST7565_NHD_C12864_4W_SW_SPI
will work. How will it look like? Can you send a picture?
You may need to adjust the contrast by setContrast.

selevo, 2020-05-11:
clear screen  with any setContrast

olikraus, 2020-05-13:
Well, ok, then use "U8X8_ST7565_ZOLEN_128X64_4W_SW_SPI"
Locate u8x8_d_st7565.c on your local disk. change the following value from 4 to 0:
https://github.com/olikraus/u8g2/blob/9ecb87c4b62e91302006d278fc3498263a6c567d/csrc/u8x8_d_st7565.c#L110

Will this help?

closed 2020-07-05 by olikraus