olikraus/u8g2 #1188: Problem using the ST7920 with esp32 using the ESP-IDF

shaharsery, 2020-05-10:
Hi, I'm trying to create a simple project with the ST7920 and the ESP32 using eclipse and the ESP-IDF but I can't get the LCD to show anything.
I've created a simple project using this [guide](http://www.lucadentella.it/en/2017/10/30/esp32-25-display-oled-con-u8g2/)
When I connect the same setup and use the Arduino IDE with the 'HelloWorld' example and 
U8G2_ST7920_128X64_F_SW_SPI everything works fine.

This is my main app:
`
#include <driver/gpio.h>
#include <driver/spi_master.h>
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <stdio.h>
#include <string.h>
#include <u8g2.h>

#include "sdkconfig.h"
#include "u8g2_esp32_hal.h"
#include "u8g2_esp32_hal.c"

#define PIN_CLK 18
#define PIN_MOSI 23
#define PIN_RESET 22
#define PIN_DC U8G2_ESP32_HAL_UNDEFINED
#define PIN_CS 5

void app_main() {
	u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	u8g2_esp32_hal.clk   = PIN_CLK;
	u8g2_esp32_hal.mosi  = PIN_MOSI;
	u8g2_esp32_hal.cs    = PIN_CS;
	u8g2_esp32_hal.dc    = PIN_DC;
	u8g2_esp32_hal.reset = PIN_RESET;
	u8g2_esp32_hal_init(u8g2_esp32_hal);

	u8g2_t u8g2; // a structure which will contain all the data for one display
	u8g2_Setup_st7920_s_128x64_f(
		&u8g2,
		U8G2_R0,
		u8x8_byte_4wire_sw_spi,
		u8g2_esp32_gpio_and_delay_cb);  // init u8g2 structure

	u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,

	u8g2_SetPowerSave(&u8g2, 0); // wake up display
	u8g2_ClearBuffer(&u8g2);
	u8g2_SetFont(&u8g2, u8g2_font_ncenB08_tr);
	u8g2_DrawStr(&u8g2, 0,15,"Hello World!");
	u8g2_SendBuffer(&u8g2);

	vTaskDelete(NULL);
}
`

I'm pretty new to all of this so I might be missing something pretty simple,
Thanks!

olikraus, 2020-05-10:
U8g2 has been developed for Arduino IDE. I can not say anything to other IDEs :-/

shaharsery, 2020-05-10:
Ok thanks, I'll look for other options

closed 2020-05-10 by shaharsery