olikraus/u8g2 #1189: 0.5 inch oled:ld7032 driver seeedstudio knockoff issues

bturner1273, 2020-05-11:
Hola! Not really an issue per say I was just wondering if we could get support for a pretty common 0.5 inch oled knock off. 

I tried to get it to work normally using every constructor but talked to the manufacturer and turns out it has a different init sequence. I ended up getting the init up and running correctly and was wondering if you could add it as an alternate init sequence.

In file u8x8_d_ld7032_60x32.c to make my init work for the knockoff I changed
```c
static const uint8_t u8x8_d_ld7032_60x32_init_seq[]
```
to
```c
static const uint8_t u8x8_d_ld7032_60x32_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */

  U8X8_CA(0x02, 0x00),		  		
  U8X8_CA(0x14, 0x00),		  		
  U8X8_CA(0x1A, 0x05),		  		
  U8X8_CA(0x1D, 0x00),		  	
  U8X8_CA(0x09, 0x00),	      		
  U8X8_CAA(0x30, 0x00, 0x3F),
  U8X8_CAA(0x32, 0x08, 0x27),
  U8X8_CA(0x34, 0x00),				
  U8X8_CA(0x35, 0x07),		
  U8X8_CA(0x36, 0x08),	
  U8X8_CA(0x37, 0x27),			
  U8X8_CA(0x38, 0x00),        
  U8X8_CA(0x39, 0x20),       
  U8X8_CA(0x10, 0x05),		  	
  U8X8_CA(0x16, 0x00),		  	
  U8X8_CA(0x18, 0x08),		  		
  U8X8_CA(0x12, 0x2F),		  	
  U8X8_CA(0x3D, 0x01),		  	
  U8X8_CA(0x3F, 0x10),		  		
  U8X8_CA(0x44, 0x02),		  		
  U8X8_CA(0x48, 0x03),		  	
  U8X8_CA(0x17, 0x00),         
  U8X8_CA(0x13, 0x01),        
  U8X8_CA(0x3F, 0x11),
  U8X8_CA(0x3D, 0x00),

  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};
```
 

[0.5 Inch OLED board Schematic.pdf](https://github.com/olikraus/u8g2/files/4611750/0.5.Inch.OLED.board.Schematic.pdf)
[0.5OLED SPEC.pdf](https://github.com/olikraus/u8g2/files/4611751/0.5OLED.SPEC.pdf)

olikraus, 2020-05-11:
Yes, could be done. I just need some time to implement this.

olikraus, 2020-12-25:
added as
``` C++
//U8G2_LD7032_60X32_ALT_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 11, /* data=*/ 12, /* cs=*/ 9, /* dc=*/ 10, /* reset=*/ 8);	// SW SPI Nano Board
//U8G2_LD7032_60X32_ALT_1_4W_SW_I2C u8g2(U8G2_R0, /* clock=*/ 11, /* data=*/ 12, /* reset=*/ U8X8_PIN_NONE);	// NOT TESTED!
```

closed 2020-12-25 by olikraus