olikraus/u8g2 #1196: SSD1322 - Right side cutoff with later library version

PeteWasEre, 2020-05-17:
After recently updating to the latest library version from a very old one (a couple of years probably, not sure which) the right hand side of my SSD1322 256x64 is cutoff. About 5% of the display is blank.

If i revert back to 2.24.3 its is better, though i still seem to have lost one character at the end of my menu which seems to have gotten slightly wider? 

Info that might help:
Constructor - U8G2_SSD1322_NHD_256X64_F_8080 u8g2(U8G2_R0, 30, 31, 32, 33, 34, 35, 36, 37, 39, 42, 40 ,41);
Device - Arduino Mega 2560

Font for the menu that seems to be stuck wider -     u8g2.setFont(u8g2_font_unifont_t_symbols);

I note some other offsets now too with fonts, havent nailed that down yet but less fussed by that. The main issue is the lost band on the right edge of the display.

PeteWasEre, 2020-05-17:
ok further info, with the reverted library its not fonts issue, there is still a small slice of the display not working. its just the slice gets much bigger beyond 2.24.3

This didnt happen before yesterday when i did a big libraries update, but not matter how far i revert the library it doesnt come back? this is trluy wierd

AnHardt, 2020-05-17:
#define U8G2_16BIT ?

PeteWasEre, 2020-05-18:
Ah sorry, forgot all about that! It works perfectly now. Thanks for the assist!

closed 2020-05-18 by PeteWasEre

olikraus, 2020-05-22:
:-)