olikraus/u8g2 #1197: Hanging other devices on the I2C bus

QuantumImage, 2020-05-19:
Hello, 
I am totally new to this so please bear with me. 
I have connected a display: 
      U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0);  //Set Display type
Onto the I2C and it works perfectly :)
I have also added a onto the I2C
    <SparkFunMLX90614.h> //Temperature Sensor I2C
And that works perfectly :)
The trouble I have is that only one of them will work at a time. Is it possible to set it up so that they both work?
My setup is:  
void setup() 
{
  Serial.begin(9600); 
  therm.begin(); 
  therm.setUnit(TEMP_C); 
  u8g2.begin();  
}
All the best
Dren

olikraus, 2020-05-19:
Hmm... what exactly is the problem? 
I mean, obviously only one device can use the bus at a time, but I assume you mean something else.

QuantumImage, 2020-05-19:
Hello Olikraus,

thank you for getting back to me so quickly.

I have a temperature sensor and display both on the I2C bus, I can get both working seperately but not together. I have a simple test program to read the temperature then display it (below).

Can you see what basic mistake I've made?

All the best

Dren

//********************
#include <Wire.h>
#include <SparkFunMLX90614.h> //Temperature Sensor I2C
#include <SPI.h>
#include <U8g2lib.h>  //OLED

U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0);  //Set Display type

IRTherm therm;

void setup()
{
  Serial.begin(9600);
  therm.begin();
  therm.setUnit(TEMP_C);
  u8g2.begin();
}
String temperature;

void loop()
{
  if (therm.read()) // On success, read() will return 1, on fail 0.
  {
    temperature = String(therm.object(), 2);
    Serial.print("Object: ");
    u8g2.drawStr(0, 20, temperature.c_str());
    Serial.println("C");
    u8g2.firstPage();
    do {
      u8g2.setFont(u8g2_font_ncenB14_tr);
      u8g2.drawStr(0,20, temperature.c_str());
     } while ( u8g2.nextPage() );
    delay(100);
  }
}

JamesGKent, 2020-05-19:
it looks like the default speed for SSD1306 is 400KHz, but the temperature sensor only supports 100KHz.

olikraus, 2020-05-19:
Could be...
See also note 2 here: https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock

QuantumImage, 2020-05-20:
Thank you for your help, I set the I2C port down to 100kHz and swapped around the temperature display begins and it all works. Excellent 😄

I appreciate your help, cheers

closed 2020-07-05 by olikraus