olikraus/u8g2 #1198: Home Button

B1Sisko, 2020-05-19:
Hi Oli, ich hänge seit tagen an diesen SelectionList Code herum und versuche mir einen Menü damit zu kreieren. Es  funktioniert alles tadellos bis auf ,das wenn ich einen Menü ausgewählt habe es nicht schaffe wieder zurück in das Hauptmenü zu gelangen. Ich betätige den Home Button der bei mir auf Pin 9 ist. aber irgendwie wird sie nicht vom Programm registriert (Außer im Hauptmenü wo sie überflüssig ist.)

ich schicke dir mal meinen code. Ich hoffe du kannst mir da weiterhelfen den ich bin langsam am verzweifeln: 

`
const int voltageSensorPin = A1;          // sensor pin
float vIn;                                // measured voltage (3.3V = max. 16.5V, 5V = max 25V)
float vOut;
float voltageSensorVal;                   // value on pin A3 (0 - 1023)
const float factor = 5.128;               // reduction factor of the Voltage Sensor shield
const float vCC = 5.00;                   // Arduino input voltage (measurable by voltmeter)

#include <U8g2lib.h>
//U8G2_SSD1327_EA_W128128_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 7, /* reset=*/ 8);
U8G2_SSD1327_EA_W128128_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 7, /* reset=*/ 8);

bool menu = 0;
uint8_t current_selection = 0;
const char *string_list =
        "VOLTMETER\n"
        "AMPERE-METER\n"
        "OHM-METER\n"
        "CONTNUITY\n"
        "ABOUT";

void setup() {
        
        
        //Serial.begin(9600);
        
        u8g2.begin(/*Select=*/ 2, /*Right/Next=*/ A3, /*Left/Prev=*/ A2, /*Up=*/ 4, /*Down=*/ 5, /*Home/Cancel=*/ 9); 

// welcome using Full screen buffer mode
        //u8g2.clearBuffer();
        
        u8g2.setFont(u8g2_font_6x13_tr);
        
        //u8g2_font_lastapprenticebold_tr 
        //u8g2.setFont(u8g2_font_tenthinguys_tu); 
        //u8g2.setFont(u8g2_font_gb16st_t_3);
        //u8g2.drawStr(30,80,"MULTIMETER MENU");
        
        //u8g2.sendBuffer();
        delay(2000);
}

void loop() {
// Page buffer mode (Picture Loop)
      
          voltageSensorVal = analogRead(voltageSensorPin);    // read the current sensor value (0 - 1023) 
          vOut = (voltageSensorVal / 1024) * vCC;             // convert the value to the real voltage on the analog pin
          vIn =  vOut * factor;                    
        
        
        if(menu==0){
        current_selection = u8g2.userInterfaceSelectionList(
                "MULTIMETER MENU",
                current_selection,
                string_list);
        }

        u8g2.firstPage();
        do {
                if ( current_selection == 0 ) {         // home/cancel button
                        menu = 0;
                }
                if (current_selection == 1) {         // VOLT
                       menu = 1;
                       draw_VOLT(); }
                        
                if (current_selection == 2) {         // RES
                       menu = 2;
                       draw_AMP();
                }
                       
                if (current_selection == 3) {         // RES
                       menu = 3;
                       draw_RES();
                }
                       
                if (current_selection == 4) {         // RES
                       menu = 4;
                       draw_ABOUT();
                }
        } while ( u8g2.nextPage() );
        
        

  delay(10);
}

void draw_VOLT(){

        //const char* VOLT = (vIn);
        
        
        //u8g2.clearBuffer();
        u8g2.setFont(u8g2_font_profont17_mr );
        u8g2.drawStr(10,18,"VOLTMETER DC");
        u8g2.drawStr(50,90,"VDC");
        u8g2.setFont( u8g2_font_logisoso30_tn); 
        u8g2.setCursor(10,65);
        u8g2.print(vIn,4);
        
        //u8g2.sendBuffer();
        
        
        //menu = 1;
        
        // Serial.print("Voltage = ");             
         //Serial.print(vIn);
         //Serial.println("V");
         
      
}       
 

void draw_RES(){
        u8g2.drawStr(5,15,"RESISTANCE Ohms");
        //u8g2.clearBuffer();
        u8g2.setCursor(20,50);
        u8g2.print("Hello World");
        //u8g2.sendBuffer();
        menu = 2;

        if ( current_selection == 0 ) {         // home/cancel button
                        menu = 0 ;
                }
  }
void draw_AMP(){
    }
  

void draw_ABOUT(){
    
  }`

olikraus, 2020-05-22:
Ohje... meine Zeit ist recht begrenzt. Also das mindeste wäre eine ordentliche formatierung des code (markdown hat da viele Möglichkeiten).
Alternativ: Wie wäre es denn mit einem ordentlichen Menü-System wie "Arduino-Menu"?

B1Sisko, 2020-05-31:
Hi habe das problem gelöst.! Habe einfach eine Button abfrage hinzugefügt in den ich die folgende Zeilen eingefügt habe. sobald ich den Homebutton (HIGH) betätige geht er wie gewünscht in das Hauptmenü.  Nicht zu vergessen es muss ein 10K Pulldown zwischen den gewünschten Digital Pin und Push Button hinzugefügt werden da sonst stark oszilliert! 

const int buttonPin = 31;     // choose a number for a pushbutton pin
 int buttonState = 0; 

 void draw_VOLT()
 
 if (buttonState == LOW) {
        menu = 1;      
       } else {
       menu = 0 ;   `

B1Sisko, 2020-05-31:
Hi habe das problem gelöst.! Habe einfach eine Button abfrage hinzugefügt. Sobald ich den Homebutton (HIGH) betätige geht er wie gewünscht in das Hauptmenü. Nicht zu vergessen es muss ein 10K Pulldown zwischen den gewünschten Digital Pin und Push Button hinzugefügt werden da es  sonst zu stark oszilliert! Die folgende Zeilen habe ich dazu verwendet.

 const int buttonPin = 31;     // choose a number for a pushbutton pin
 int buttonState = 0; 

 void draw_VOLT()
 
 if (buttonState == LOW) {
        menu = 1;      
       } else {
       menu = 0 ;   
       }

B1Sisko, 2020-05-31:
Hi habe das problem gelöst.! Habe einfach eine Button abfrage hinzugefügt. Sobald ich den Homebutton (HIGH) betätige geht er wie gewünscht in das Hauptmenü. Nicht zu vergessen es muss ein 10K Pulldown Resistor zwischen den gewünschten Digital Pin und Push Button hinzugefügt werden da es sonst zu stark oszilliert! Die folgende Zeilen habe ich dazu verwendet.

const int buttonPin = 31; // choose a number for a pushbutton pin
int buttonState = 0;

void draw_VOLT()

if (buttonState == LOW) {
menu = 1;
} else {
menu = 0 ;
}

olikraus, 2020-06-05:
ah, ok :-)

closed 2020-06-06 by B1Sisko

dalwinder1578singh, 2020-06-06:
Organic Audio clip