olikraus/u8g2 #1200: LX-12864B11 from FYSETC AIO II printer board. works in Marlin (U8Glib) but not with baremetal U8G2

sl1pkn07, 2020-05-20:
Hi i have this printer board https://wiki.fysetc.com/AIO_II/

with have a LX-12864B11 display

<details>

[image]

</details>

works as spected with Marlin firmware, wich use the library u8glib as lcd display driver

but i want to use with this project https://github.com/technik-gegg/SMuFF-1.1, wich use U8G2 library for LCD. but the LCD not works, the LCD do nothing

you can see the details in https://github.com/technik-gegg/SMuFF-1.1/pull/11 , also ypou can see what i test, including the U8G2 demos

greetings

olikraus, 2020-05-22:
I need a more specific question... not sure how I can help here.

sl1pkn07, 2020-05-29:
Hi

basically i've used all U1701 and ST7565 constructors (SPI) and none works with this display, Brand engineer claims this display comes with this controller. but not works with u8g2. but this board, with Marlin firmware, with this controller (see the latest comments in smuff PR , which the engineer talk about what constructor is used in marlin), works perfect with u8glib

[image]

greetings

olikraus, 2020-06-05:
U8g2 supports all the u8glib displays. Usually there is a wiring issue, I mean, did you use the correct arguments to the u8g2 constructor?

sl1pkn07, 2020-06-05:
this can't be a wiring issue because all is integrated in the PCB

the code i use is:

~~~
[platformio]
default_envs      = test_fysect

[common]
lib_deps          =
  u8g2
build_flags       =
build_unflags     = -W unused-variable

[env:test_arduino]
platform          = atmelavr
framework         = arduino
board             = megaatmega2560
build_flags       = ${common.build_flags}
                    -std=gnu++14
                    -g
                    -D__BDR_ARDUINO
;                    -D__WS_I2C_SW
                    -D__WS_I2C_HW
build_unflags     = -std=gnu++11 $(common.build_unflags)
lib_ldf_mode      = chain
lib_deps          = ${common.lib_deps}

[env:test_nucleo]
platform          = ststm32
framework         = arduino
;board             = nucleo_f103rb
board             = genericSTM32F103RB
build_flags       = ${common.build_flags}
                    -std=gnu++14
                    -g
                    -D__BDR_NUCLEO
                    -D__WS_I2C_SW
;                    -D__WS_I2C_HW
build_unflags     = -std=gnu++11 $(common.build_unflags)
lib_ldf_mode      = chain
lib_deps          = ${common.lib_deps}
upload_protocol   = stlink
debug_tool        = stlink

[env:test_fysect]
platform          = ststm32
framework         = arduino
board             = genericSTM32F103RC
build_flags       = ${common.build_flags}
                    -std=gnu++14
                    -g
                    -D__BRD_FYSETC
build_unflags     = -std=gnu++11 $(common.build_unflags)
lib_ldf_mode      = chain
lib_deps          = ${common.lib_deps}
extra_scripts     = FYSETC_AIOII.py
upload_protocol   = serial
debug_tool        = stlink
~~~

~~~
from os.path import join
from os.path import expandvars
Import("env", "projenv")

# Custom HEX from ELF
env.AddPostAction(
    join("$BUILD_DIR","${PROGNAME}.elf"),
    env.VerboseAction(" ".join([
        "$OBJCOPY", "-O ihex", "$TARGET", # TARGET=.pio/build/FYSECT_AIOII/firmware.elf
        "\"" + join("$BUILD_DIR","${PROGNAME}.hex") + "\"", # Note: $BUILD_DIR is a full path
    ]), "Building $TARGET"))

# please keep $SOURCE variable, it will be replaced with a path to firmware

# In-line command with arguments
UPLOAD_TOOL="stm32flash"
platform = env.PioPlatform()
if platform.get_package_dir("tool-stm32duino") != None:
    UPLOAD_TOOL=expandvars("\"" + join(platform.get_package_dir("tool-stm32duino"),"stm32flash","stm32flash") + "\"")

env.Replace(
    UPLOADER=UPLOAD_TOOL,
    UPLOADCMD=expandvars(UPLOAD_TOOL + " -v -i rts,-dtr,dtr -R -b 115200 -g 0x8000000 -w \"" + join("$BUILD_DIR","${PROGNAME}.hex")+"\"" + " $UPLOAD_PORT")
)
~~~

~~~
#include <U8g2lib.h>

/*
U8g2lib Example Overview:
Frame Buffer Examples: clearBuffer/sendBuffer. Fast, but may not work with all Arduino boards because of RAM consumption
Page Buffer Examples: firstPage/nextPage. Less RAM usage, should work with all Arduino boards.
U8x8 Text Only Example: No RAM usage, direct communication with display controller. No graphics, 8x8 Text only.

This is a frame buffer example.
*/

#ifdef __BDR_NUCLEO
  #ifdef __WS_I2C_SW
    #define DSP_SCL           PC0   // (A5) SCL
    #define DSP_SDA           PC1   // (A4) SDA
  #endif
#endif

#ifdef __BRD_FYSETC
  #define DSP_CS_PIN          PB5
  #define DSP_DC_PIN          PA15
  #define DSP_RESET_PIN       PB4

  #define RGB_LED_R_PIN       PB0
  #define RGB_LED_G_PIN       PB6
  #define RGB_LED_B_PIN       PB7
#endif

/* Constructor */
#if defined(__BDR_NUCLEO) || defined(__BDR_ARDUINO)
  #ifdef __WS_I2C_HW
    U8G2_SSD1327_WS_128X128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
  #elif __WS_I2C_SW
    U8G2_SSD1327_WS_128X128_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ DSP_SCL, /* data=*/ DSP_SDA, /* reset=*/ U8X8_PIN_NONE);
  #endif
#elif __BRD_FYSETC
  U8G2_UC1701_MINI12864_F_2ND_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ DSP_CS_PIN, /* dc=*/ DSP_DC_PIN, /* reset=*/ U8X8_PIN_NONE);
#endif

//#define MINI_LOGO

void setup(void) {
  #ifdef __BRD_FYSETC
    pinMode(RGB_LED_R_PIN, OUTPUT);
    pinMode(RGB_LED_G_PIN, OUTPUT);
    pinMode(RGB_LED_B_PIN, OUTPUT);

    digitalWrite(RGB_LED_R_PIN, 1);
    digitalWrite(RGB_LED_G_PIN, 1);
    digitalWrite(RGB_LED_B_PIN, 1);
  #endif
  Serial.begin(115200);
  u8g2.begin();
}

void drawLogo(void) {
  u8g2.setFontMode(1);  // Transparent
  #ifdef MINI_LOGO
    u8g2.setFontDirection(0);
    u8g2.setFont(u8g2_font_inb16_mf);
    u8g2.drawStr(0, 22, "U");

    u8g2.setFontDirection(1);
    u8g2.setFont(u8g2_font_inb19_mn);
    u8g2.drawStr(14,8,"8");

    u8g2.setFontDirection(0);
    u8g2.setFont(u8g2_font_inb16_mf);
    u8g2.drawStr(36,22,"g");
    u8g2.drawStr(48,22,"\xb2");

    u8g2.drawHLine(2, 25, 34);
    u8g2.drawHLine(3, 26, 34);
    u8g2.drawVLine(32, 22, 12);
    u8g2.drawVLine(33, 23, 12);
  #else
    u8g2.setFontDirection(0);
    u8g2.setFont(u8g2_font_inb24_mf);
    u8g2.drawStr(0, 30, "U");

    u8g2.setFontDirection(1);
    u8g2.setFont(u8g2_font_inb30_mn);
    u8g2.drawStr(21,8,"8");

    u8g2.setFontDirection(0);
    u8g2.setFont(u8g2_font_inb24_mf);
    u8g2.drawStr(51,30,"g");
    u8g2.drawStr(67,30,"\xb2");

    u8g2.drawHLine(2, 35, 47);
    u8g2.drawHLine(3, 36, 47);
    u8g2.drawVLine(45, 32, 12);
    u8g2.drawVLine(46, 33, 12);
  #endif
}

void drawURL(void)
{
#ifndef MINI_LOGO
  u8g2.setFont(u8g2_font_4x6_tr);
  if ( u8g2.getDisplayHeight() < 59 )
  {
    u8g2.drawStr(89,20,"github.com");
    u8g2.drawStr(73,29,"/olikraus/u8g2");
  }
  else
  {
    u8g2.drawStr(1,54,"github.com/olikraus/u8g2");
  }
#endif
}

void loop(void) {
  u8g2.clearBuffer();
  drawLogo();
  drawURL();
  u8g2.sendBuffer();
  delay(1000);
}
~~~

all builded with platformio+vscode 

(U8G2_UC1701_MINI12864_F_2ND_4W_HW_SPI is the latest of the constructors i've tested)

the board scheme

https://github.com/FYSETC/AIO_II/blob/master/AIO%20II%20V3.2%20SCH.pdf

greetings

olikraus, 2020-06-11:
hmmm... from your could you seem to select the I2C display (BDR_ARDUINO) however the schematic connects to an SPI display.
This can not work. You should recheck whether your constructor and the corresponding args fit to your hardware.

olikraus, 2020-06-11:
Oh, there are multiple build flags. ok, that means your info is still very much incomplete... Nevertheless: What I wrote before is still valid.

sl1pkn07, 2020-06-11:
Hi

what you mean with incomplete? the code is taked from https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_full_buffer/U8g2Logo/U8g2Logo.ino and use the constuctors from here https://github.com/olikraus/u8g2/wiki

the code works with Arduino Mega and Nucleo_F103RB board (both with Waveshare 1.5" oled I2C display (hardware i2c, with software i2c in the nucleo board (not tested in the arduino, as you can see in teh code)) have problems with the display render))

greetings

olikraus, 2020-06-11:
> what you mean with incomplete? 

The information and the code you provided is confusing. How shell we know which build flags are used? I can not know this. How will I know which ifdefs in your code are used? I do not know this. And even if I would have a lot of time, i can not figure this out, because I do not know which  build flags are active. 

So the question is:
Did you use this constructor:
    U8G2_SSD1327_WS_128X128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
or this
    U8G2_SSD1327_WS_128X128_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ DSP_SCL, /* data=*/ DSP_SDA, /* reset=*/ U8X8_PIN_NONE);
or this?
  U8G2_UC1701_MINI12864_F_2ND_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ DSP_CS_PIN, /* dc=*/ DSP_DC_PIN, /* reset=*/ U8X8_PIN_NONE);

I do not know this. You need to tell which constructor you have used.

Anyhow, if i assume, that you have used 
  U8G2_UC1701_MINI12864_F_2ND_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ DSP_CS_PIN, /* dc=*/ DSP_DC_PIN, /* reset=*/ U8X8_PIN_NONE);

then it looks correct regarding the schematic, except for the reset line, which is not connect.
In fact the missing reset line could be the problem. You should tell the pin for the reset line as last
argument to the constructor.

sl1pkn07, 2020-06-11:
try with vscode, have syntax highligth, and use platformio as trigger. the build flags is the default used by platformio

olikraus, 2020-06-11:
What is vscode? what is platformio? 
You wouldn't belive, but this is issue number 1200. So there had been >1000 questions from other users. There are sooo many platforms, IDEs, code editors and so on available. I can not know them all and i can definitly not support them.
If vscode/platformio decided to use u8g2, then please also approach those people regarding u8g2.
I have created u8g2 for the Arduino IDE and for the standard Arduino boards.
It is far beyond my available time and resources to support anything else.

This is open source. This is has been developed in my spare time. I did this, because it had been fun for me. But it also means, you can not expect me to try anything else or know anything else. I am sure vscode/platformio is super cool and great, but I simply do not have the time to look at this...

As a result: 
 - I can not help on any other platform except Arduino IDE
 - I can not review your hardware and wiring
 - I can not review your code
 - I can only try to give you generic advices
 - I can point out how to write software with u8g2 
 - I am happy to receive bug fix reports regarding u8g2 as a library
 - I need a precise problem description: What is wrong? What is instead expected?

If this is not acceptable for you, it is probably better to search for professional support.
Again: This is open source. You do not need to pay anything here. 

:-)

sl1pkn07, 2020-06-11:
is not works in in platformio, not works is arduino IDE, no is not a platfoimio/vscode problem

as i say, the code works in Arduino mega and nucleo (STM32F103RB chip) with I2C display (hardware I2C), but not the Fysetc board (STM32F103RCT6 chip) with SPI LX-12864B11 display. wich this last one working with Marlin firmware, wich use platformio and you u8glib2 library (as you can see in the photo posted before)

so i assume is problem of u8g2 library

i' povide the code, i provide the build environment, i provide the hardware schemas, i use the same example you wrote in the examples

As a result:

> I can not help on any other platform except Arduino IDE

you not need add or fix other platform. yust the library

> I can not review your hardware and wiring

you can see the schemas i provide. is a comercial product and definitely works with you old u8glib library. so is a regresion

>I can not review your code

all the code i use as example is posted in https://github.com/olikraus/u8g2/issues/1200#issuecomment-639343816, all of them, no more, nor less (u8g2 library sources provided by platformio https://platformio.org/lib/show/942/U8g2 wich use the latest release in this repository)

>I can only try to give you generic advices

i apreciate it, but for this, i open the issue, because not work, and IDK how fix it. i'm not programmer/coder

>I can point out how to write software with u8g2

my code (really is you code, is a copy paste of you examples) is for test the hardware, the real code is in https://github.com/technik-gegg/SMuFF-1.1/tree/SMuFF-2.0-(Experimental) and works, but not in this hardware. you can talk with @technik-gegg

>I am happy to receive bug fix reports regarding u8g2 as a library

is a u8g2 problem. works in older u8glib library (marlin use it), but not in new u8g2 library

>I need a precise problem description: What is wrong? What is instead expected?

the problem is the display not works, not init, not do things, not show nothing with u8g2, but works with u8glib. the expected is the display works/init, show the u8g2 logo in the display, like do the I2C display i tested in other hardware

greetings

olikraus, 2020-06-11:
> (really is you code, is a copy paste of you examples)

Maybe it was my code once, but it got modified a lot.

> is a u8g2 problem

It it is a u8g2 I would ask you to exactly point out the problem. Saying it does not work is not enough. There are many reasons why something does not work.

As I said before, the missing reset line argument could be a problem.

sl1pkn07, 2020-06-11:
Hi

is not hard modified, real code is untouched 

`drawlogo()` `drawurl()` and `loop()` is untouched. `setup()` i only add the LCD (RGB, set in white) backlit pins (always ON)

if you mean this
~~~
/ reset=*/ U8X8_PIN_NONE
~~~

i have tested with this one and with PB4, wich is the reset pin in the display

[image]
[image]

in both cases, not works in all constructors

greetings

olikraus, 2020-06-11:
Did you check whether the signals arrive at the LCD?

sl1pkn07, 2020-06-11:
how? i don't have oscilloscope, or you mean with multimeter?

technik-gegg, 2020-06-12:
@olikraus 
Hi Oli,
first of all, let me thank you for creating such a beautiful library. I've used it in various projects with various displays, so far without any issue. I have to confess, it sometimes is hard finding the right constructor, since there are so many of them and not all displays are documented well.
Though, this is not the fault of the library. 

As far as it concerns my project, I don't believe it's some U8G2 bug here. 
So, far I've compared the init sequence given by the FYSETC engineer that's used in Marlin and there is little to no difference to the init sequence used in the U8G2 library. 
I didn't do any testing on the real hardware myself yet, due to lack of time and ressources. I will eventually but there's no time frame yet.
As a conclusion, I'd recommend to close this bug report for now, since I'm pretty confident that it has nothing to do with the U8G2 library. I may come back to you, once I had the chance to test this setup.

@sl1pkn07 
Judging from your picture of the display running the test code, I'd say the display has turned on all pixels. If that's the case, you'll never see any characters / images on the display.
I know from experience that not all displays initialize the contrast at the same level. You ought to try setting the contrast of your display to a different value in the setup() (i.e. setContrast(127)) or, even better, start from 0 and increment the contrast in steps of 5 the main loop after the delay(). This way you'll be able to tell whether or not the display is reacting to the commands sent. If the commands are received and interpreted correctly, you will at least see that the pixels are going from light to dark over time.

olikraus, 2020-06-12:
> how? i don't have oscilloscope, or you mean with multimeter?

@sl1pkn07 Sometimes an oscilloscope or logic analyzer is required to debug the hardware and ensure that everything works.

@technik-gegg 
Thanks for your support here. Checking the contrast is indeed a good idea.

sl1pkn07, 2020-06-12:
something like this? (idea taked from http://samsneatprojectblogcode.blogspot.com/2016/06/piezo-buzzer-code-and-fritzing.html)

~~~
void loop(void) {
  u8g2.clearBuffer();
  drawLogo();
  drawURL();
  u8g2.sendBuffer();
  delay(1000);
  int count[26]={5, 10, 15, 20, 25, 30, 35, 40, 45, 50, 55, 60, 65, 70, 75, 80, 85, 90, 95, 100, 105, 110, 115, 120, 125, 127};
  for(int i = 0; i < 26; i++){
    u8g2.setContrast(count[i]);
  }
}
~~~

?

technik-gegg, 2020-06-12:
More like:

```
int contrast = 5;
bool dir = false;

void loop(void) {
  u8g2.setContrast(contrast);
  u8g2.clearBuffer();
  drawLogo();
  drawURL();
  u8g2.sendBuffer();
  delay(500);
  if(contrast >= 250 && !dir)
   dir = true;
  if(contrast <=5 && dir)
   dir = false;
  if(!dir)
    contrast += 5;
 else 
    contrast -= 5;
 Serial.print("Contrast: ");
 Serial.println(contrast);
}
```
This way it'll cycle up and down and also send the current value to the serial monitor.
If you, at some point, see something on the display, write down the contrast value shown in the serial monitor and set this in the setup() routine.

sl1pkn07, 2020-06-12:
Hi

tested and not works, the serial monitor (putty on ttyusb0) show the contrast values, but the display show nothing. always black

greetings

technik-gegg, 2020-06-13:
Well, the easiest way to find the right initialisation for the display is this (at least, that's how I do it usually):

- Open Arduino
- load the example **U8g2 -> full_buffer -> GraphicsTest**
- setup the pins for DC_PIN and CS_PIN with a **#define** at the very top of the program
- remove the comment (**//**) in front of the most probable constructor ending with **_128X64_F_HW_SPI**
- set (..., /\* cs=\*/ CS_PIN, /\* dc=\*/ DC_PIN, /\* reset=\*/ U8X8_PIN_NONE) within the constructor
- compile and upload the example - let it run for a couple of seconds and see what you get on the display
- if you get nothing, put back the comment sign (**//**) and move on to the next most probable constructor

At some point you should at least be able to see something on the screen. If you're lucky enough, you may see the graphics test running without issues. 
In some cases you may discover either interlace lines, or the display not being fully populated or the graphics being cut off. This means, you're close but it's still not the right constructor - try a similar one.

olikraus, 2020-06-13:
Yes, but also note, that not all constructors are listed in the examples. 
Complete list is here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp

technik-gegg, 2020-06-14:
@sl1pkn07 
Something caught my eye browsing the [Marlin code for the AIOII](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/stm32f1/pins_FYSETC_AIO_II.h):

Line 33 says: 
```
#define pins_v2_20190128 
```
And according to this, lines 144-148 say:
```
    #ifdef pins_v2_20190128
      #define DOGLCD_CS                     PB5
    #else
      #define DOGLCD_CS                     PB7
    #endif
```
Further, lines 175-177 say:
```
  #ifndef RGB_LED_B_PIN
      #define RGB_LED_B_PIN                 PB7
  #endif
```
Whereas your test code says: 
```
#ifdef __BRD_FYSETC
  #define DSP_CS_PIN          PB5
  #define DSP_DC_PIN          PA15
  #define DSP_RESET_PIN       PB4

  #define RGB_LED_R_PIN       PB0
  #define RGB_LED_G_PIN       PB6
  #define RGB_LED_B_PIN       PB7
#endif
```
This raises the assumption that you're using the **pins_v2_20190128** configuration in your example. I'm not sure about the reason for this definition but I assume there have been changes in the PCB wiring.

So, what if your display doesn't operate on this **pins_v2_20190128** configuration? 

In this case the **DSP_DC_PIN** would be **PB7**, which - in your sample - is used for the **RGB_LED_B_PIN**. That might explain why you're not seeing anything on the display. 
Try swapping pins **DSP_CS_PIN** (=PB7)  and **RGB_LED_B_PIN** (=PB5) in your test code and run it again. 
BTW: The Marlin code also states that there is a LCD Reset pin, which is either on **PB4** or **PB6**. You may try this out as well.

sl1pkn07, 2020-06-14:
i have already test the RGB colours, blue light is PB7, and also, in the configuration.h for this board (with i have used in my board, as-sis):

https://github.com/MarlinFirmware/Configurations/blob/f1d378e8e064d03d17e9c61712be609d258e83ce/config/examples/FYSETC/AIO_II/Configuration.h#L2228-L2233
(https://github.com/sl1pkn07/Marlin/blob/d11b81541780ab41da1b69b82c1bb81808efaa49/Marlin/Configuration.h#L2224-L2229)

technik-gegg, 2020-06-14:
@sl1pkn07 
There's one last thing you can try:

The schematics for the AIO II say that the **DOGLCD-A0** signal (which translates to **DSP_DC_PIN** in your test code) is shared with the JTAG pin **JTDI**. 

As far as I know, on the STM32 the JTAG is enabled by default because it's being used for the STLink program/debug functions, which means, you can't use one of this pins for I/O by default. 
Hence, you have to disable the JTAG feature before you can reconfigure/use anyone of these pins. 

This also correlates to the statement in line 31 of the Marlin code I was referring to in my previous post.
```
#define DISABLE_JTAG
```

To do so, you have to call the STM32 library function **afio_cfg_debug_ports(AFIO_DEBUG_NONE)** somewhere in your **setup()** routine. This will turn off the JTAG and free this pin for the display. 

**Please be aware that this call will not magically turn on the display and make everything work.**
It'll just free the pin, so it can be used for the display. You still have to find the right constructor for your display.

sl1pkn07, 2020-06-14:
Just put that line in the first line in `setup()` and set the contrast to 250, and set properly the reset pin (not tested without it):

[image]

without contrast i can see the graphics, but only if view in angle, so i play with it and seems 230 is a good value

[image]

@technik-gegg you are a Hero, @olikraus thank you for all. and very, very very sorry to both

this issue finaly can be close.

gretings and again, thanks

closed 2020-06-14 by sl1pkn07

olikraus, 2020-06-14:
I am happy to see that finally everything works. :-)

@technik-gegg : Excellent research

technik-gegg, 2020-06-14:
@olikraus 
Danke für das Kompliment!

sl1pkn07, 2020-12-19:
Hi again. since then, i have no problems with the display, but yesterday, just discover my display, LX-12864B11, works with this two contructors

U8G2_UC1701_MINI12864_F_4W_HW_SPI
U8G2_ST7567_JLX12864_F_4W_HW_SPI

the first one, U8G2_UC1701_MINI12864_F_4W_HW_SPI, needs ser the rotation as U8G2_R0, and set the contrast to 230 (by code)
the last one, U8G2_ST7567_JLX12864_F_4W_HW_SPI, need set the rotation as U8G2_R0, and set the contrast to 180 (by code) because 230 used in U8G2_UC1701_MINI12864_F_4W_HW_SPI is too bright

my question is, what of the two constructors is the correct for the display?. works both, but i dont't know what is the consequence for use the "wrong" one

greetings

olikraus, 2020-12-21:
> my question is, what of the two constructors is the correct for the display?. works both, but i dont't know what is the consequence for use the "wrong" one

There are no consequences and probably almost no difference except the contrast value.
With "by code" you mean the setContrast (https://github.com/olikraus/u8g2/wiki/u8g2reference#setcontrast) command?

sl1pkn07, 2020-12-21:
i use the board/lcd with a project wich use u8g2, and idk how works at all, i only need edit one #define, wich set the contrast. needing lower value for U8G2_ST7567_JLX12864_F_4W_HW_SPI, and upper for U8G2_UC1701_MINI12864_F_4W_HW_SPI (and set the rotation direction in the constructor)... so yes, maybe use that comand

olikraus, 2020-12-21:
ah, ok :+1: