olikraus/u8g2 #1201: I2C Optimization

billwatersiii, 2020-05-20:
Instead of having a function in u8x8_cad.c micromanage the I2C transaction (start, stop, etc), I would like u8g2 to use a function that I provide that does the complete I2C transaction.  Without modifying the code in u8x8_cad.c, is there any way to do this?

v1993, 2020-05-20:
I think your best bet is to allocate a buffer and store data there. IIRC, my ESP32 driver does something like that: https://github.com/v1993/u8g2/blob/master/sys/esp-idf/U8x8lib.c#L365

olikraus, 2020-05-22:
Just write your own byte function. See the porting guide.

billwatersiii, 2020-05-22:
Ah.  Thanks!

closed 2020-05-22 by billwatersiii

olikraus, 2020-05-22:
Here is the link: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-i2c-communication

billwatersiii, 2020-05-22:
Right.  I found that, and got it all working.  Thanks again.