olikraus/u8g2 #1202: HelloWorld.ino sketch not working with ESP32, SSD1306 display, Arduino IDE

butchalline, 2020-05-20:
Board, display and wiring proven OK with Adafruit libraries and Arduino IDE, but no luck with U8G2 .  Here is my "skinny" version of the HelloWorld sketch:

#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
// I have tried many U8G2_SSD1306_128X64...HW_I2C u8g2() versions, but the above 
// gives the most readable display. See attached photo.

void setup(void){
  u8g2.begin();
}
void loop(void){
  u8g2.clearBuffer();          // clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr); // choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");  // write something to the internal memory
  u8g2.sendBuffer();          // transfer internal memory to the display
  delay(1000);  
}
Here is what is displayed with the above sketch.  Hello W flashes in the upper left corner of display and the remainder of the display is static.

[image]

Any and all help appreciated - thanks

butchalline, 2020-05-20:
Update: By accident, I power cycled the USB port that the ESP32 was attached to and the display showed Hello World just like it is supposed to do!! 
Stay tuned for deeper investigations as to why this happened...

laci200270, 2020-06-02:
Maybe check if your screen has a broken corner..
That caused issues with mine

closed 2020-07-05 by olikraus