olikraus/u8g2 #1204: Overflow on u8g2.drawRBox

Algorab, 2020-05-25:
If height is less than height based of the radius of two opposite corners a box overflow is happen.
Example for overflow: u8g2.drawRBox(88, 150, 30, 9, 5);

Would expecting that box is working till height 0 on radius 5 for example.

olikraus, 2020-06-05:
The height includes the the radius. How should the box look like if the hight is lesser then twice the radius? Such a frame does not make sense.

See also the restrictions mentioned here: https://github.com/olikraus/u8g2/wiki/u8g2reference#drawrbox

The behavior of the RBox is just not defined under your assumption.

closed 2020-07-05 by olikraus

thisdp, 2025-08-24:
It does make sense when radius is larget than width/height, especially for progressbar or other animations. During the animation, it will enter that condition, and the whole screen just renders wrong pixels.