olikraus/u8g2 #1207: u8g2 examples

jbltimberrock, 2020-05-26:
I just installed the u8g2 library and was trying to load and run Examples. None of the examples will compile without this error: 'u8g2' was not declared in this scope. Not even Hello World.  I'm using a UNO R3. Shouldn't examples run as-is without any other effort? Any suggestions?

jbltimberrock, 2020-05-27:
Further, there appears to be at least two sketches in the Examples therein that DO NOT conform to Arduino naming protocol as messages appear when starting Arduino about sketches beginning with numbers. I believe Library Mgr said Version 2.27.2

jbltimberrock, 2020-05-29:
Better off to use Adafruit_SSD1306 and Adafruit_GFX. They are foolproof.

closed 2020-05-29 by jbltimberrock

olikraus, 2020-06-05:
> I just installed the u8g2 library and was trying to load and run Examples. None of the examples will compile without this error: 'u8g2' was not declared in this scope. 

You need to uncomment the display as adviced in the comments of the examples.

jbltimberrock, 2020-06-07:
that's OK, I'll use Adafruit libraries and have no issues...I've already
deleted u8g2