olikraus/u8g2 #1211: Inserting converted custom font.

alienclay, 2020-06-01:
I'm a novice with c but I am completely inexperienced in altering libraries. How do I take a custom font file (converted from .bdf to .c ) and use it in my Arduino sketch. Also, It would be easier to use the lighter 8x8 than u8g2 but I'll take a method for either.

AnHardt, 2020-06-01:
Starting from a TrueTypeFont *.ttf use [otf2bdf](https://github.com/olikraus/u8g2/tree/master/tools/font/otf2bdf) to make a *.bdf font.
Then use [bdfconv]([url](https://github.com/olikraus/u8g2/tree/master/tools/font/bdfconv)) to produce a *.h file containing the font.
Include the *.h file into your sketch.
Switch to the font with [setfont](https://github.com/olikraus/u8g2/wiki/u8g2reference#setfont) to use it. 

If you are using u8g2 or 8x8 should not matter, but for 8x8 the fonts are restricted to be 8x8 (or multiples of that?).

olikraus, 2020-06-05:
> Include the *.h file into your sketch.
You could also just paste the generated c code (it is just an array) into the .ino file.

> If you are using u8g2 or 8x8 should not matter, but for 8x8 the fonts are restricted to be 8x8 (or multiples of that?).

Almost true, but u8x8 font format is different and a different commandline option is require for bdfconv.c

closed 2020-07-05 by olikraus