olikraus/u8g2 #1212: Issue in RA8835 320x240 initialization

dyezepchik, 2020-06-01:
Found this library so cool! Thanks! Got my winstar 320x240 screen working in minutes with a generic stm32f103rb board and arduino IDE.
But then noticed that I can draw only in a 240x240 rectangle. After 240 pixels by X axis all the graphics overflows to the beginning of the screen i.e. 0px by X. I'm using U8G2_RA8835_320X240_1_6800 constructor. 
 @olikraus , could you please point me what to look at?

dyezepchik, 2020-06-01:
sorry, it was 16 bit mode

closed 2020-06-01 by dyezepchik

olikraus, 2020-06-11:
:-)

Hmmm... maybe I should activate 16 bit once u8g2 compiles for __ARM__

dyezepchik, 2020-06-12:
Well, actually it depends not on ARM, but on what we need to address in the screen. If the screen by at least one axis is more than 256, probably it's needed? I don't remember how exactly my screen memory is addressed, was connecting it long time ago...

olikraus, 2020-06-13:
true, but I can not change the code at runtime from 8 to 16 bit. I could just enable 16 for all displays on those platforms which used to have more flash and RAM.

Let's see. I am happy that your problem was solved.