olikraus/u8g2 #1213: Help: Advice required: DOGXL160w

ravimiranda, 2020-06-03:
@olikraus 
I am at a crossroad here, and would like to know if the library can be used on  stm32f103rct6 blue pill platform? Are there any examples? I also have a DOGL128 that I would like to experiment with on the stm32f103rct6 blue pill, can the current library be used to program the stm32f103rct6 blue pill using the Arduino IDE?

I also noticed that there is a raspberry pi version that is available for the DOG displays, will this work over I2C and a raspberry pi zero?

Many thanks for your time.
Ravi

olikraus, 2020-06-05:
U8g2 has been written for and tested with standard Arduino boards. Third party "so called Arduino compatible" boards might be supported, but basically I can not promise anything.

I heard about success stories for blue pill with u8g2, but I personally was not able to use blue pill boards.

ravimiranda, 2020-06-08:
@olikraus 
Thank you for your reply. Could I ask if there is a tutorial that demonstrates the use of this library on raspberry pi? I might really not have the expertise of porting the library to STM32xx.

Many thanks.

Ravi

olikraus, 2020-06-11:
> Could I ask if there is a tutorial that demonstrates the use of this library on raspberry pi? I might really not have the expertise of porting the library to STM32xx.

There are two different approaches depending on the question, whether there is a Arduino board description and an integration into the Arduino IDE available. If this is the case, then u8g2 should work in almost all cases.

For the STM32xx there should be an integration available, so I see a good chance to use u8g2.
Probably google for STM32 and Arduino IDE

If there is not Arduino IDE integration available, then maybe the u8g2 porting guide might help. 
This is true for the Raspberry Pi, but I would claim that this requires a certain skill level and is not supported by me: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

closed 2020-07-05 by olikraus