olikraus/u8g2 #1214: u8g2 hangs on begin

vpeppers, 2020-06-04:
I am working on a project using an Arduino Nano Every.  The project is for a speedometer, and has many moving pieces.  I have two I2C OLED displays, using a TCA9548a mutliplexer.  I also have a Hall effect sensor to generate pulses.  I have been testing and verifying the code.  I recently started testing the Hall Effect sensor, and suddenly my displays quit working.  I tracked the problem down to the ug82.begin() command.  The program hangs at that.  I have built a test program and it hangs in that, too.  

<code>
#include <Wire.h>
#include <U8g2lib.h>

U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0);

void setup() {
  Serial.begin(115200);
  TCA9548A(0);
  Serial.println("TCA9548A");
  u8g2.setI2CAddress(0x3c);
  u8g2.begin();
  Serial.println("u8g2 begin");
 }

void loop() {  
} 

void TCA9548A(uint8_t bus)  {
  Wire.beginTransmission(0x70);  // TCA9548A address is 0x70
  Wire.write(1 << bus);          // send byte to select bus
  Wire.endTransmission();
}
</code>

I have no idea where to go from here.  Do you?

dalwinder1578singh, 2020-06-05:
> #include #include

I have recrded audio file for evry one. Do not wory about the language it
is sound source. If you listen one channel at atime nothing looks
suspeceius in both channels audio level remains same. When i play with
stereo headphone audio source moves laft right and center.
Check it out and let me know.thanks

olikraus, 2020-06-05:
@dalwinder1578singh not sure how this could help for this topic
@vpeppers I2C is not equal to I2C. Maybe there is a bus speed problem: See note 2 here: https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock

vpeppers, 2020-06-05:
I tried both higher and lower bus speeds, as well as lowering the processor speed fro 20MHz to 16MHz.  I even disconnected the Hall Effect sensor.  No change.  Please note that it WAS working.  I have no idea what might have changed that would have affected the initialization of the u8g2 library.

dalwinder1578singh, 2020-06-05:
It is going to hange because you did not listen organic audio clip. Here is
one listen with stereo headphone and find out how audio source moves left
right and center. Thanks

XFer012, 2020-06-05:
vpeppers, maybe the display was faulty and now no longer works.
It occurred to me once (0.96" no-name white OLED).

olikraus, 2020-06-06:
If u8g2 was working and does not work any more after removal of the other I2C device, then something is broken in your setup.
@dalwinder1578singh : This is a project regarding electronics. Please contribute technical suggestion.

dalwinder1578singh, 2020-06-06:
Some times it happen brokem i2c transmission end when you connect one more
device with same i2c address you will wounder both will work.
Check it out my Organic audio clip
First listen each channel one by one lastly listen both channel at same
time with steteo headphone you will discover sound source moves. That is
it. Thanks

vpeppers, 2020-06-18:
It was a bad mux.  I replaced that, and everything works.  Thanks for addressing the issue so promptly.

closed 2020-06-18 by vpeppers