olikraus/u8g2 #1215: Missing method to set OLED brightness with U8x8Log

XFer012, 2020-06-05:
Hello,
I'm using U8x8Log with an SSD1306 OLED, very nice, thanks for this excellent library!
I would like to raise the brightness because I need to work outside and the tiny screen is hardly readable; but I did not find a method to to that, at least on the reference page:
https://github.com/olikraus/u8g2/wiki/u8logreference

Is there a way?

(I'm using: 
U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE);
....
U8X8LOG g_u8x8log;
)

Thanks

XFer012, 2020-06-06:
Ok, I've found that I have to set contrast with the u8x8 object, not with the u8x8log object, so it's 

u8x8.setContrast(255)

Unfortunately it's still very dim (maybe because I only have Vcc=3.3v available, not 5v), but it's not a library shortcoming

closed 2020-06-06 by XFer012

olikraus, 2020-06-11:
Maybe also your power supply is also not able to deliver more current. 
An additional cap between gnd and power supply may also help.

XFer012, 2020-06-12:
No, it works fine with high brightness if I use the regular U8g2 (not the U8x8log).
I ended up using the regular U8g2, it was unreadable otherwise.

olikraus, 2020-06-13:
ok... strange...