olikraus/u8g2 #1216: setContrast not working on Nokia 5110 display

ChrisXenon, 2020-06-07:
Oli, I am in awe of the contributions you have made - thank you.
I am driving a Nokia 5110 from a Wemos D1 mini. Everything works fine except that setContrast() has no effect, and whilst I have found a couple of other people saying similar none relate to this situation, so I hope you can help.

The code snippet is:

 if   (rightButtonPressed)
       {displayContrast = displayContrast + 25;
        if (displayContrast > 255) displayContrast = 0;
        u8g2.setContrast(displayContrast);
        rightButtonPressed = false;
        return;
       };

Do I need to do something else for the contrast change to take effect? A sendBuffer() is done subsequent to this elsewhere, but it doesn't help.

Many thanks
Chris

olikraus, 2020-06-11:
Actually set contrast will execute the following code:
https://github.com/olikraus/u8g2/blob/99b63a38a3fde09a453fe0143c567bfa242d1c09/csrc/u8x8_d_pcd8544_84x48.c#L124-L128

It should actually work.

ChrisXenon, 2020-06-12:
Thank you for your reply Oli. Well, it doesn't work. Perhaps I have a faulty unit. I ordered 3 more and they arrived but none work. It seems the zebra strip and netallic frame mechanism for connecting the PCB to the display is very unreliable. Will your library work on the HX1230 display? This unit bonds the display to the PCB and has better resolution.

Thanks again.
Chris

olikraus, 2020-06-13:
HX1230 support is there: https://github.com/olikraus/u8g2/issues?q=is%3Aissue+hx1230

ChrisXenon, 2020-06-13:
Apologies - I'm wasting your time.

Thanks again
Chris

olikraus, 2020-06-13:
Not a problem. Other topics require much more effort ;)

closed 2022-03-04 by olikraus