olikraus/u8g2 #1217: drawFrame height is not correct!

dnrelectronica, 2020-06-08:
hi all, 

I use the DrawFrame function and found out filling in the height is wrong with a offset of 1 pixel. 
For example if you fill in a height of 10 , the actual height which is drawn on the screen is 11 pixels.
If you draw a Box of height 10, the height is correct and is indeed 10. Why is there a difference in height between the Box and Frame? Is there something wrong with the calculation of the borders in the frame when calculating height? border is 1 px. 

Kind Regards, 

Corne

olikraus, 2020-06-11:
I have tested this:
https://github.com/olikraus/u8g2/blob/99b63a38a3fde09a453fe0143c567bfa242d1c09/sys/sdl/issue1217/main.c#L33-L37

Actually I can not find any difference in the height. So I can not reproduce your problem.

dnrelectronica, 2020-06-12:
Hi Olikraus, 

Thank you for taking the time to dive into this issue. In the meantime i found out the issue is not there if i use the library in my real application. But i posted this issue when i was designing my screen with a u8g2 simulator :

olikraus, 2020-06-12:
Ah, ok. Thanks for the feedback.

closed 2020-06-12 by olikraus