olikraus/u8g2 #1219: Persian letters Problem

ahmadelwaly2016, 2020-06-11:
when i try to write Arabic words the letters be **separated**.
that's my code , can any one help me to solve this problem.

`#include <Adafruit_ILI9341.h>

Adafruit_ILI9341 tft(A5, A3, A4); //CS, DC, RST
#include <U8g2_for_Adafruit_GFX.h>

U8G2_FOR_ADAFRUIT_GFX u8g2_for_adafruit_gfx;

//define some colour values

#define TFT_BLACK   0x0000
#define TFT_WHITE   0xFFFF
#define TFT_BLUE    0x001F
#define TFT_RED     0xF800
#define TFT_GREEN   0x07E0
#define TFT_CYAN    0x07FF
#define TFT_MAGENTA 0xF81F
#define TFT_YELLOW  0xFFE0

void setup()
{
  tft.begin();
  u8g2_for_adafruit_gfx.begin(tft);                 // connect u8g2 procedures to Adafruit GFX
  u8g2_for_adafruit_gfx.setFontDirection(0);       // left to right
  if (tft.width() < tft.height()) tft.setRotation(1);
  u8g2_for_adafruit_gfx.setFontMode(0);                 // use u8g2 transparent mode (this is default)
  u8g2_for_adafruit_gfx.setForegroundColor(TFT_BLACK);  // apply Adafruit GFX color
  u8g2_for_adafruit_gfx.setBackgroundColor(TFT_YELLOW);  // apply Adafruit GFX color
}

char Elwaly[] = "الوالي ";
char num[] = "الشاشه ترحب بكم " ;

void UTF8_rt_lft(int x, int y, const char *u8str)
{
  char buf[5], c, n;
  while (c = *u8str) {
    switch (c & 0xF0) {   //escape sequences
      case 0xC0: n = 2; break;
      case 0xD0: n = 2; break;
      case 0xE0: n = 3; break;
      case 0xF0: n = 4; break;
      default: n = 1; break;
    }
    memcpy(buf, u8str, n); //copy one glyph at a time
    buf[n] = 0;   //terminate with NUL
    u8str += n;
    uint8_t len = u8g2_for_adafruit_gfx.getUTF8Width(buf);
    x -= len;     //calc new posn for glyph
    u8g2_for_adafruit_gfx.drawUTF8(x, y, buf);
  }

}

void showmsgs(const uint8_t *font, int x, int y)
{

                                  
  u8g2_for_adafruit_gfx.setFont(u8g2_font_ganj_nameh_sans16_t_all);

  
  UTF8_rt_lft(x, y += 30, Elwaly);
  UTF8_rt_lft(x, y += 20, num);

}

void loop()
{
  int x = tft.width();                                // write from right
  tft.fillScreen(TFT_YELLOW);

  showmsgs(u8g2_font_ganj_nameh_sans16_t_all, x, 10);
#ifndef AVR
  showmsgs(u8g2_font_ganj_nameh_sans16_t_all, x, 80);
  showmsgs(u8g2_font_ganj_nameh_sans16_t_all, x, 160);
#endif
  delay(2000);

}`

olikraus, 2020-06-11:
which problem?

ahmadelwaly2016, 2020-06-12:
when i write those words
 char Elwaly[] = "الوالي ";
char num[] = "الشاشه ترحب بكم " ;

displayed in tft with **separated** letters .

AnHardt, 2020-06-12:
Have you had a look into: "[Including support for Arabic languages](https://github.com/olikraus/u8g2/issues/640)" ?

idreamsi, 2020-06-13:
https://github.com/idreamsi/u8g2-persian-reshaper

closed 2020-06-16 by ahmadelwaly2016