olikraus/u8g2 #1220: Persian letters

ahmadelwaly2016, 2020-06-11:
how i can use those fonts with 20 pt or 24 pt ?

https://github.com/rastikerdar/samim-font
Sizes: 10pt, 12pt, 14pt, 16pt

https://github.com/rastikerdar/samim-font
Sizes: 10pt, 12pt, 14pt, 16pt
https://github.com/font-store/GanjnamehFont
Sizes: 10pt, 12pt, 14pt, 16pt

https://github.com/nekofar/iranian-sans-fontface
Sizes: 8pt, 10pt, 12pt, 14pt, 16pt

olikraus, 2020-06-11:
You need to convert the font to the bdf font file format. Once this is done, you can use the u8g2 tool bdfconv.exe to convert the bdf font file to the internal format.
The fonts mentioned above seem to be available in .ttf format. 
Starting with this there are two options for the conversion to bdf:
 * otf2bdf, see here: http://sofia.nmsu.edu/~mleisher/Software/otf2bdf/
 * fontforge, see here: https://learn.adafruit.com/custom-fonts-for-pyportal-circuitpython-display/conversion

You need a bdf file for each font size, this means you will have 17 font files at the end.

ahmadelwaly2016, 2020-06-14:
https://fonts2u.com/download/ge-ss-two-bold.font this is the font with bitmap

olikraus, 2020-06-14:
The zip file does not contain a .bdf file.
It does not make sense to send me anything else. I really need a .bdf file.

olikraus, 2020-06-14:
This is what is required: https://en.wikipedia.org/wiki/Glyph_Bitmap_Distribution_Format

ahmadelwaly2016, 2020-06-15:
this is the bdf , [](https://drive.google.com/file/d/1cKTov_uxB5jC-dWAhrt6j-XsTsyR2LlI/view?usp=sharing) . Excuse me from my lack of experience in this.

ahmadelwaly2016, 2020-06-15:
https://drive.google.com/file/d/1cKTov_uxB5jC-dWAhrt6j-XsTsyR2LlI/view?usp=sharing

olikraus, 2020-06-21:
I have used the following command to convert the font:
```
./bdfconv -v -f 1 -m '$$20-$$ffff,~1588,~65010,~65205,~65206' GESSTwoBold-Bold-75.bdf -o u8g2_font_gess.c  -n u8g2_font_gess -d ../bdf/helvB18.bdf 
```
I actually had to exclude several glyphs like unicode 1588, 65010, 65205 and 65206 which are actually too big for u8g2. Indeed there is a kind of a max size in u8g2 and it seems, that you font actually hits that max size.

On the other side, your glyphs is really big. For example unicode 1586 has a height of 60 pixel. Unicode 1587 has a width of 90 pixel. Is this really what you want? You will be able to show only one or two glyphs on a 128x64 display.

Anyhow, the font is here. You can copy and past this font into your code. Below, I attached an overview picture of the font.

```
const uint8_t u8g2_font_gess[23100] U8G2_FONT_SECTION("u8g2_font_gess") = 
  "\35\0\6\6\7\7\4\7\10oi\370\341\60\0\64\362\0\0\0\0\7H \7\0\0\6\37\1!\35"
  "\13\230\2%\201\77\340\24(\202\204\220\334\200\30\16\304q\205\241(\30\305\351\77\340\5\42H\24\212~"
  "\61\1\304\20\31\20\1b@\4\10\342\20 \210Cp(\16\301\241\70\4\207\342\20\34\212\203`(\16"
  "\202\241\70\10\206\342 \30\12\303`\60\12\203\301(\14\6\243\60\24\216\302P\70\12C\341(\20\204\243"
  " \0\0$\204(\340\322\134\245\302X|!\202\213P\227g\303x\222\220!*DF\310\20\31\14\242"
  "@ \216EQ\60\12\310\242($\1da\24\244\205Q\220\30EAj\20\205\351\21\24\246\312\224y"
  "\70\215g\371(_\213\207\351<\231\311\42Y\244\2\321\220\12\5C*\30\13\251`,\244\202\261(\4"
  "K\301X\20\205\244P\60\6\304Q \32Bc(\10\216\241$\224\4\223P*D\310g\331@\26\353"
  "\341\25\306\342/\2\0'\23\10\212~\31\201WpL\60~Ba(\42\10\2\0(N\25\341\312\64"
  "!\201D\200\14\16bI\60\10&\261$[\24KbQ,\211E\261$\26\305\222X\224\226\304\242L"
  "Q,\312\377\337\262X\224\61\11F\261(\230\4\243`\22\214\202I\60\12&\321$\61\12&\321$k"
  "\16OA\22P\201\1\0)L\26\241\312\64\211\0\65\220\14\17\242Q\256I\70I\215\242I\64\212&"
  "\321(\232D\243`\24\215\202Qj\24\214\362\230\305\262`\224\337\242`\224\307$\30%&\301(\61\11"
  "F\301$\61\12&\251I\60\311\65\10\347\0\25\244\1\25\210\0*;\226\212\6\67\215A!X\10\5"
  "\1\61\14\6\244\60\24\222\202PX\12A\201I\64\210\343\7h\216\203\223`\12Aa)\4\5\245\60"
  "\24\20\303`\70\12E!\31\24\3F\320\0\0+\27%\222\22S\233\205\266\374_\363\3\376\3$\315"
  "B[\376_\3\0-\12\245\204\62S\201\377\1\22.\12\11\205\2\35\201\37\260\1/M\36\330\2G"
  ")\305\244\240\24T\202RP\11JiRP\212IA)\250\4\245\64)(\305\244\240\24T\202R\232"
  "\24\224bRP\12*A)M\12J\61)(\5\225\240\224&\5\245\230\24\224\202JPJ\223\202R"
  "L\12JA%*\0\60\32\15\311\37\71\1\301BH\16F\202X\10\77 \65(\12\4RP\6L"
  "\0\61\24\12\230\3-\223\0\42\60\10Ea\70\10\77\340\177\200\14\62,\37\330\3Y\235\310BI,"
  "\310\345\300\30\231BCl\6\316\60!\10\303\3jHL\323\202RT\12jA)*\345\377\377\377U"
  "\0\63I+\230\3o\1\5\222\60(H\302\240 \11\203\202$\14\12\222\60(H\302\240 \11\203\202"
  "$\14\312\241\60(G\302\250\34\11\243b(\12\13\301\30\32AFx\4)\1\65\234\10SA:D"
  "\211\247Q\204\312\377\377\177!\0\64D \230\3Y\25JrA\60G\306\320\24\33\202kpH\14\212"
  "A-\252\345\254E\265\260\30\324\344\211@\20\311\1A\34I\63a(\13\203Z\252\226\257bP\214\252"
  "\61}\4\257\241Cp\212\315\221I.\14\5\65Y\244\224\17a\241\301w\340\24\31\303zP\21i\2"
  "U\234\13\223Y\26\3#Y\24\214c\351P$\213b\241,\10\305\262\30\24\315BP\64\213`\341("
  "\2\305S\201\250\357\251<\212`\341(\4E\263\20\26\13cX$\14\302\251\360\30\35d\243d\32\313"
  "\63\231\36Z\1\2\0\66+\37\330\3W\1\310\63a*\312\5\301\34\31CSh\212M\61($\222"
  "cbP\14j\251ZP\213J\271jA-\377\377\377\1\67z$\230\3_\1\305ci\26\301\242Y"
  "\10\212f!(\232\205\240h\24\303bY\20\212eA(\226\5\241X\24\305BQ\30\212da(\222"
  "\205\241H\24\307\2Q\356X \212C\221(\16\205\242\60\24\212\302P(\212B\261(\12\5\243 \24"
  "\214\202P\60\12\42\321(\6\205\223\30\24\216BP\70\12!\361(\204\4\224\10\24\20\25\64\221\244I"
  "\21\12\32\365\260\234Y\15\214\301\0\0\70\177#\230\3_\227\206\325\260\34\225\203z\242 \23\224\24\221"
  "$\221\24\64\201(\17\345I\4\12G!$\34\205\220p\24\202\242I\14\12F\61(\30\5\221`\24"
  "\204bI\24\12EQ(\24E\241P\24\206\42Q\30\12Dq(\20\305\241@\24\307\342P \12C"
  "\221(\14E\242\60\24\211\302P$\13b\241(\10\305\242 \24\213\202P,\13a\261,\4\5\263\20"
  "\24\215BP\64\226\246\342T\234\5\71C#\230\3_\27F\25\201*\315e\311$\233\243S|\210\317"
  "\20\22\32G#h(\230\345\322X\71\325k.\314ea\4\14\205\21\70\14G \22\302\6\20\361)"
  "<F\347\340(\216@\321 \10E\246\374\77:\17\211\215\2\35\201\37\260\371\1\300\17\330\0<\17\235"
  "\321\2I#\6\304\374\377\223\230\377\177=\21%\314\36W\201\377\1\222\37X\370\3\376\3$>\21\235"
  "\321\2I\1\206\304\374\377\203\230\377\177\21\0[\25\220\237\322(\201\37\260\62\224\377\377\377\377\377\377\31"
  "\77 \2\134U\36\230\2G\1E\245\240\24U\242RT\211JA)*\5\245\250\22\225\242JT\12"
  "JQ)(E\225\250\24U\242RP\212JA)\252D\245\250\22\225\202RT\12JQ%*E\225"
  "\250\24\224\242RP\212*Q)\252D\245\240\24\225\202RT\211JQ%\0]\25\220\237\322(\201\37"
  "\20:\222\377\377\377\377\377\377\35\77@\3_\12\32D\336:\201\177\300\4{\66\31\341\326@\261\300\243"
  "`\32\12\203A@H\21\2BD\310\377\377 D\204\200\220=\211(!%\242D\204\210\22\21\362\377"
  "\177Q\42BD\211h\361@\26G\204\0|\13\207\340\326\32\201\377\7\70\0}\63\231\340\326@\1\305"
  "\323x\30\22\262\10)J@\311\377_\204LBNJ\70\211'\31\204\210\20P\362\377\27!A\210\10"
  "\351I\70\13\344\241\64\230\203\4\0\240\7\0\0\6_\1\253V\25\216\2\65\217\200\22@\206\214\42\302"
  "P\30\6Ca \12F\201\60\24\206\301P\30\10Ca \12F\241(\24\6\302P\30\10Ca "
[... cut ...]
  ",\22\5\262H\24\311\2Q$\13D\221,\20\205\262\70\24\312\342P(\13c\241,\214\305\262(\26"
  "\313\202`,\213\341\7\374\202R@\22F\303h\10\25 \243\223\0\0\376\374|\63X\3m;E\252"
  "\234\263\200\24\317\342Q<\213G\361,\36\305\263x\24\320\302Q@\13G\1-\34E\264h\24\321\242"
  "QD\213F\21-\32\205\264`\24\322\202QH\13F\61-\26\305\264X\24\323bQL\213EA-"
  "\24\5\265P\24\324BQT\12EQ-\22E\265H\24\325\42QX\13Da-\20\205\265\70\26\227"
  "\342X\134\13cq-\212\5\266 \30\315\17\370\177\200\1\376\377\10\0\0\6\1\1\0";
```
[image]

ahmadelwaly2016, 2020-06-21:
Thank You vey much oliver for your help.

ahmadelwaly2016, 2020-06-21:
can you tell me the way to make a code for a font like you

olikraus, 2020-06-21:
> can you tell me the way to make a code for a font like you

You have to execute bdfconv:
```
./bdfconv -v -f 1 -m '$$20-$$ffff,~1588,~65010,~65205,~65206' GESSTwoBold-Bold-75.bdf -o u8g2_font_gess.c  -n u8g2_font_gess -d ../bdf/helvB18.bdf
```

bdfconv is available here: https://github.com/olikraus/u8g2/tree/master/tools/font/bdfconv

Also consider the FAQ:
https://github.com/olikraus/u8g2/blob/99b63a38a3fde09a453fe0143c567bfa242d1c09/doc/faq.txt#L233-L251

closed 2020-07-05 by olikraus