olikraus/u8g2 #1221: Adding a second ST7920 display - interference (noise) problems with u8g2.begin();

vmoct, 2020-06-11:
Adding a second display (specifically adding the second begin() call) causes visual interference on the second display, using either software SPI or hardware SPI (same symptoms). The problem moves with the code - it is not hardware noise. Individual instances work fine and parallel displays (same instance) work fine. Using SS with pin 10 (default) may be a problem. Great library otherwise so far.

U8G2_ST7920_128X64_1_HW_SPI u8g22(U8G2_R1,9,7);
U8G2_ST7920_128X64_1_HW_SPI u8g2(U8G2_R1,10,8);
//U8G2_ST7920_128X64_1_SW_SPI u8g22(U8G2_R1,13,11,9,7);
//U8G2_ST7920_128X64_1_SW_SPI u8g2(U8G2_R1,13,11,10,8);

void setup(void) {

  u8g2.begin();         //adding a second begin() causes issues (noise) on other display
  u8g22.begin();  
}

int num = 0;

void loop(void) {

  u8g22.firstPage();
  do {
      postNumber22(9-num);          //draw a graphical 7 segment digit (uses RBox)
  } while ( u8g22.nextPage());

  u8g2.firstPage();
  do {
      postNumber2(num);             //draw a graphical 7 segment digit (uses RBox)
  } while ( u8g2.nextPage());
  
  delay(2000);
  if (++num > 9) num = 0;
}

dalwinder1578singh, 2020-06-11:
Not corelate with my project

olikraus, 2020-06-11:
hmm... I do not see a mistake in the code. How do the visual interference look like?

vmoct, 2020-06-11:
Thanks for the very quick reply. Here are a couple of pictures. Adding a
second instance usually affects only the display with the second called
begin(). However often both displays are affected with added horizontal
(vertical using 90R) lines and areas overlapping or moved (SPI sync
problems?). The example shown just put up a single digit in the biggest
font.

olikraus, 2020-06-11:
> Here are a couple of pictures. 

ahm... where exactly?

AnHardt, 2020-06-12:
Using two ST7920 on the same SPI-bus is impossible without interference because of a hardware limitation of the chips. http://www.hpinfotech.ro/ST7920.pdf page 26 states:
> When chip select (CS) is low, ST7920 serial clock counter and serial data will be reset. Serial transfer counter is set
to the first bit and data register is cleared. After CS is “L”, any further change on SID or SCLK is not allowed. It is
recommended to keep SCLK at “L” and SID at the last status before set CS to “L”. For a minimal system with only
one ST7920 and one MPU, only SCLK and SID pins are necessary. CS pin should pull to high. 

That means CS is broken.
Any SPI-traffic to any other device does disturb the display even when CE is not active.
It needed several iterations of the datasheet until they admitted that that clearly.

vmoct, 2020-06-12:
Thank you for the detailed explanation and information. I did get it
working properly with one HW SPI and one SW SPI instance and it is fast
enough for my purposes. The u8g2 library is quite impressive, especially
with the hardware support provided. Thanks again. VM

closed 2020-07-05 by olikraus