olikraus/u8g2 #1223: LoadFromSD.ino is not working as expected

e-labInnovations, 2020-06-13:
Code: [LoadFromSD.ino](https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/LoadFromSD/LoadFromSD.ino)
Board: Arduino UNO
OLED: `U8G2_SSD1306_64X32_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);`
SD Card Module: [HW-125](https://file.vishnumaiea.in/hardware/micro-sd/Catalex-Micro-SD-Card-Module_3_LS.jpg)
Output Serial Monitor:
[image]
Outupt OLED:
[image]

olikraus, 2020-06-13:
Which display did you connect? Is this really a 64x32 display?

e-labInnovations, 2020-06-14:
128x64

olikraus, 2020-06-14:
Nice.

> Can I use .xbm file instead of .bin file?

Yes, but you need to write a .xbm format reader instead of the current drawFile() procedure.
https://github.com/olikraus/u8g2/blob/99b63a38a3fde09a453fe0143c567bfa242d1c09/sys/arduino/u8g2_page_buffer/LoadFromSD/LoadFromSD.ino#L377

I decided to use a special binary format to speed up the reading process. .xbm will slowdown loading and increase code in your uC.

e-labInnovations, 2020-06-14:
**Thank you so much**

How can I convert my `.xbm` image file to `.bin` file?
I think png2bin tool can create a `.bin` file from a `.png` file. Can you give example command of png2bin tool?

olikraus, 2020-06-14:
> How can I convert my .xbm image file to .bin file?

The code is included in the .ino file. On the other side you could use "gimp" to convert xbm to png.

The command will be:
```
png2bin picture.png picture.bin
```
This will generate picture.bin out of picture.png.

closed 2020-06-15 by e-labInnovations