olikraus/u8g2 #1227: GME128128-02 128x128 OLED pixel rows lost

ChrisXenon, 2020-06-19:
Hey Oli, sorry to bother you again so soon.

Because I can't get setContrast() to work on the Nokia 5110 display with u8g2 I'm trying another display, this time an OLED with "GME128128-02" writtn on its back. It uses the SSD1327 driver chip.

I have one hooked up to an ESP-based Wemos D1 mini via I2C.  It works, but the pixel rows below row 95 do not display. The example code below shows the square frame properly except the bottom lines are missing. From other test programs, I know I cannot write text to the lower lines either.

If I measure the physical width of the displayed 128 x 128 frame it is 26mm.
If the physical HEIGHT was the same it would have to display off the bottom of the display.
This leads me to suspect that pixel (0,0) is some way down the actual display - not at its top,
and this would account for the lost space at the bottom.

I've got two of these displays and both behave the same way.

There is no constructor in your full list which mentions "GME" so I used another which works but has this "lost height" problem. I've tried all the qualifying constructors (F, 128x128, HW, I2C) - and those that work all fail in this way. Is there a specific one I should use?

Just getting a small functional display is proving far harder than I ever expected.

Anyway, any help would be appreciated.

EXAMPLE PROGRAM:

#include <Arduino.h>
#include <U8g2lib.h>
#include <ESP8266WiFi.h>
#include <Wire.h>

U8G2_SSD1327_EA_W128128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

void setup() {
  u8g2.begin(); u8g2.setContrast(50);
   u8g2.clearBuffer();
   u8g2.drawFrame(0, 0, 127, 127);
    u8g2.sendBuffer();
}

void loop() {}

olikraus, 2020-06-21:
> This leads me to suspect that pixel (0,0) is some way down the actual display - not at its top,
and this would account for the lost space at the bottom.

So you mean, that the display is somehow shifted?

You could try to change the second value of this line to something else
https://github.com/olikraus/u8g2/blob/99b63a38a3fde09a453fe0143c567bfa242d1c09/csrc/u8x8_d_ssd1327.c#L498

So you need to locate u8x8_d_ssd1327.c on your local system and change the second argument.
Maybe try `U8X8_CA(0x0a1, 32),` which should shift by 32 rows.

When building, ensure that the Arduino IDE recompiles u8g2 sources.

olikraus, 2020-07-05:
no feedback... closing...

closed 2020-07-05 by olikraus

ChrisXenon, 2020-07-05:
Sorry Oli, I saw your reply, and became despondent about the entire p-roject, and then forgot it.
Thanks for your support.

olikraus, 2020-07-06:
No prolem... I just need to clean up this issue list once in a while. Please reopen if you have further questions.

zentT, 2020-07-10:
Hello,

Thank you for your library.
I have the same screen labelled "GME128128-02" at its back. I also have a 
[Mega pro CH340G ATmega2560](https://robotdyn.com/pub/media/0G-00005641==MEGA-PRO-CH340GATmega2560/DOCS/PINOUT==0G-00005641==MEGA-PRO-CH340GATmega2560.jpg)
I'd like to understand what's the difference between SH and HW. I'm trying to use HW because you said it's much faster but I can't enter the right pin in the constructor. Also it seems like HW is made for Arduino Uno and not Mega, so I don't know what to do.
Could you point me the best constructor to use for my application ?

My goal is to monitor some values with the fastest rendering possible with my setup.
Is u8x8 much faster than u8g2 ?

Thank you very much.

olikraus, 2020-07-12:
> I'd like to understand what's the difference between SH and HW. 

HW (hardware) SPI is faster because it will use Arduino SPI library (SPI.h). Arduino SPI is connected to specific pins, which are different for each board. You need to figure out the SPI (MOSI and SCK) pins for your board and connect your display accordingly.
As a consequence you do not need to name those pins as args for the U8g2/U8x8 Constructor, because they a fixed by your Hardware and the Arduino SPI lib.

SW (software) SPI emulates the SPI protocol with GPIO ports: You just provide all the required pin names to the u8g2/u8x8 constructor.

> Is u8x8 much faster than u8g2 ?

U8g2 is based on U8x8. U8x8 is the low level interface with lesser functionality. But yes, it will be faster, use lesser RAM and lesser code.

zentT, 2020-07-13:
Thank you very much for your answer. I got HW right but the problem is I can't get U8g2 to work with HW. 

This one is working good : U8G2_SSD1327_EA_W128128_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 21, /* data=*/ 20, /* reset=*/ U8X8_PIN_NONE);
While the HW version doesn't work at all : 
U8G2_SSD1327_EA_W128128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

Am I missing something or HW isn't compatible with my screen ? 

Also I need a fast rendering (<0.5 sec), do you think it's achievable with my setup (ATMega2560) ? 

Thank you

zentT, 2020-07-13:
Also how am I suppose to complete this : 
U8X8_SSD1327_MIDAS_128X128_HW_I2C([reset [, clock, data]]);
The [[,, makes it difficult to understand. Why should I put the SDA and SCL pins if it's HW ? 

Thanks,

Térence

dannield, 2022-01-24:
First off, thanks for this post, it pointed me to the right direction.
i changed lines 460-580 to look like this:
/*=============================================*/
/*  EA W128128 round OLED 128x128 */
/* issue #641 */
/* https://www.lcd-module.de/fileadmin/eng/pdf/grafik/W128128-XR.pdf */

static const u8x8_display_info_t u8x8_ssd1327_ea_w128128_display_info =
{
  /* chip_enable_level = */ 0,
  /* chip_disable_level = */ 1,
  
  /* post_chip_enable_wait_ns = */ 20,
  /* pre_chip_disable_wait_ns = */ 10,
  /* reset_pulse_width_ms = */ 100, 	
  /* post_reset_wait_ms = */ 100, 		/**/
  /* sda_setup_time_ns = */ 100,		/* */
  /* sck_pulse_width_ns = */ 100,	/*  */
  /* sck_clock_hz = */ 4000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
  /* spi_mode = */ 0,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 1,	/* use 1 instead of 4, because the SSD1327 seems to be very slow */
  /* data_setup_time_ns = */ 40,
  /* write_pulse_width_ns = */ 60,	
  /* tile_width = */ 16,
  /* tile_hight = */ 16,
  /* default_x_offset = */ 0,
  /* flipmode_x_offset = */ 0,		
  /* pixel_width = */ 128,
  /* pixel_height = */ 128
};

/* this is a copy of the init sequence for the seeed 96x96 oled */
static const uint8_t u8x8_d_ssd1327_ea_w128128_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  
  U8X8_CA(0x0fd, 0x012),		/* unlock display, usually not required because the display is unlocked after reset */
  U8X8_C(0x0ae),		                /* display off */
  //U8X8_CA(0x0a8, 0x03f),		/* multiplex ratio: 0x03f * 1/64 duty */
  //U8X8_CA(0x0a8, 0x05f),		/* multiplex ratio: 0x05f * 1/64 duty */
  U8X8_CA(0x0a8, 0x07f),		/* multiplex ratio: 0x05f * 1/64 duty */
  U8X8_CA(0x0a1, 0x000),		/* display start line */
  //U8X8_CA(0x0a2, 0x04c),		/* display offset, shift mapping ram counter */
  
  U8X8_CA(0x0a2, 0x000),		/* display offset, shift mapping ram counter */
  U8X8_CA(0x0a0, 0x051),		/* remap configuration */
  
  
  U8X8_CA(0x0ab, 0x001),		/* Enable internal VDD regulator (RESET) */
  //U8X8_CA(0x081, 0x070),		/* contrast, brightness, 0..128 */
  U8X8_CA(0x081, 0x053),		/* contrast, brightness, 0..128 */
  //U8X8_CA(0x0b1, 0x055),                    /* phase length */
  U8X8_CA(0x0b1, 0x051),                    /* phase length */  
  //U8X8_CA(0x0b3, 0x091),		/* set display clock divide ratio/oscillator frequency (set clock as 135 frames/sec) */			
  U8X8_CA(0x0b3, 0x001),		/* set display clock divide ratio/oscillator frequency  */			
  
  //? U8X8_CA(0x0ad, 0x002),		/* master configuration: disable embedded DC-DC, enable internal VCOMH */
  //? U8X8_C(0x086),				/* full current range (0x084, 0x085, 0x086) */
  
  U8X8_C(0x0b9),				/* use linear lookup table */

  //U8X8_CA(0x0bc, 0x010),                    /* pre-charge voltage level */
  U8X8_CA(0x0bc, 0x008),                    /* pre-charge voltage level */
  //U8X8_CA(0x0be, 0x01c),                     /* VCOMH voltage */
  U8X8_CA(0x0be, 0x007),                     /* VCOMH voltage */
  U8X8_CA(0x0b6, 0x001),		/* second precharge */
  U8X8_CA(0x0d5, 0x062),		/* enable second precharge, internal vsl (bit0 = 0) */
  
  U8X8_C(0x0a4),				/* normal display mode */
    
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

static const uint8_t u8x8_d_ssd1327_ea_w128128_flip0_seq[] = {
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_CA(0x0a2, 0x000),		/* display offset, shift mapping ram counter */
  U8X8_CA(0x0a0, 0x051),		/* remap configuration */
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

static const uint8_t u8x8_d_ssd1327_ea_w128128_flip1_seq[] = {
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_CA(0x0a2, 0x000),		/* display offset, shift mapping ram counter */
  U8X8_CA(0x0a0, 0x042),		/* remap configuration */
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

uint8_t u8x8_d_ssd1327_ea_w128128(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  if ( u8x8_d_ssd1327_96x96_generic(u8x8, msg, arg_int, arg_ptr) != 0 )
    return 1;
  if ( msg == U8X8_MSG_DISPLAY_SETUP_MEMORY )
  {
    u8x8_d_helper_display_setup_memory(u8x8, &u8x8_ssd1327_ea_w128128_display_info);
    return 1;
  }
  else if ( msg == U8X8_MSG_DISPLAY_INIT )
  {
    u8x8_d_helper_display_init(u8x8);
    u8x8_cad_SendSequence(u8x8, u8x8_d_ssd1327_ea_w128128_init_seq);    
    return 1;
  }
  else if  ( msg == U8X8_MSG_DISPLAY_SET_FLIP_MODE )
  {
    if ( arg_int == 0 )
    {
      u8x8_cad_SendSequence(u8x8, u8x8_d_ssd1327_ea_w128128_flip0_seq);
      u8x8->x_offset = u8x8->display_info->default_x_offset;
    }
    else
    {
      u8x8_cad_SendSequence(u8x8, u8x8_d_ssd1327_ea_w128128_flip1_seq);
      u8x8->x_offset = u8x8->display_info->flipmode_x_offset;
    }
    return 1;
  }
  return 0;
}

home it helps some one in the future....