olikraus/u8g2 #1228: Allow negative height for drawFrame and drawBox

alka79, 2020-06-20:
Hello,
First, thanks for the great job with u8g2 :) I use it with SSD1306 128x64 OLED. Never used OLED before and it works just smooth.

This is a suggestion, not an issue.

A typical use case for drawBoxe is to draw bars, like a bar chart. The bars are usually on top of a baseline. It is a twist in mind to draw a bar chart with a (left,top) reference point for each bar. 
Another typical use case is to draw a box to clear an area of the display. The Glyph or text print functions use left bottom as a reference. 

I know we can dot it with some offset manipulation in our code, but the lib could make it easier for us.
I would suggest to allow negative height. This would simply mean "draw the frame or box the other vertical direction", ie upwards in most cases.

For consistency, you may also consider negative width, eventhough in my (short) experience the need is less frequent.

Thanks for considering,
Al

olikraus, 2020-06-21:
I agree to your application, but I usually try to keep the size as small as possible: Users who do not want this minus feature otherwise would have to deal with additional code which would be useless for them.
Instead I suggest to create small helper functions (not tested): 
``` C++
void drawFlexBox(U8G2 &u8g2, u8g2_uint_t x, u8g2_uint_t y, u8g2_uint_t w, u8g2_uint_t h)
{
   if ( h < 0 ) { y+=h; h = -h; }
   if ( w < 0 ) { x+=w; w = -w; }
   u8g2.drawBox(x, y, w, h);
}
```

alka79, 2020-06-21:
Thanks for the quick reply.

I did something similar in my code to offset the coordinates but your way of doing it with a helper function is much nicer. 
danke sehr,
Al

closed 2020-06-21 by alka79