olikraus/u8g2 #1229: Question:  Any plans to support SSD1351

Subaru4WD, 2020-06-23:
I have a project I would like to build and the code uses U8G2 for the display.    The display I bought has a SSD1351 and I would like to use it with this project.   Is there any plan to support the SSD1351 128x128 from Adafruit?

Thanks in advance,

olikraus, 2020-06-24:
U8g2 is a library for monochrome displays. SSD1351 is a true color display instead, so it is not supported.
However, if you like the u8g2 fonts, there is Adafruit GFX add on, which adds the font engine from u8g2 to this display: https://github.com/olikraus/U8g2_for_Adafruit_GFX

Subaru4WD, 2020-06-24:
Okay thank you for your reply.    I just want to be able to use this display on a project but that project uses the U8G2 library.

olikraus, 2020-06-24:
Maybe a SH1107 display could be used: https://buyzero.de/products/1-12-mono-oled-128x128-white-black-breakout-i2c#

Subaru4WD, 2020-06-25:
Thank you for the suggestion.   I was trying to complete this project without having to buy another display.   I purchased this display and then realized it had the SSD1351 controller and now I am trying to find a workaround so I can use it.

olikraus, 2020-06-28:
I assume this can be closed...

closed 2020-06-28 by olikraus