olikraus/u8g2 #1231: SPI conflict with U8G2_SH1107 and Adafruit_BMP3XX

alnitac, 2020-06-26:
Hello,
I'm struggling since several days with this problem.
I have a Pimoroni Oled 1"12 display and one BMP388 connected to an Arduino Iot33.
Separately both are working fine, but together there is no way.
Both are on hardware SPI, CS 9 and 10.
With bmp.begin() first, the display works fine, but no pressure measurement.
With u8g2.begin() first, pressure measurement works fine, but no display.
Someone an idea how i can resolve this ?
Thanks.

olikraus, 2020-06-26:
You probably need to reduce the spi clock speed. See the notes in the u8g2 reference manual for setBusClock.

alnitac, 2020-06-26:
Clock speed is 66 kHz when display works, and 1 Mhz when BMP works.
Setting clock speed in u8g2 doesn't change anything.
I need to investigate the BMP library to find a solution.
Thanks for the guidance.

alnitac, 2020-06-28:
I've tried to change the SPI clock speed in both librairies with no better result. 
I continue to investigate, found in BMP3XX library that SPI_MODE0 is not defined in SPI.beginTransaction, don't know if this is important.

olikraus, 2020-06-28:
> I continue to investigate, found in BMP3XX library that SPI_MODE0 is not defined in SPI.beginTransaction, don't know if this is important.

The default should be spi mode 0 and also the SH1107 driver will use mode 0, so this should not be a problem.

https://github.com/olikraus/u8g2/blob/0966f30f7daeb9249f1a4d2042460932a0b2ac48/csrc/u8x8_d_sh1107.c#L368

BTW: Which constructor did you use? You must use the HW_SPI constructor.

alnitac, 2020-06-29:
Ouch, i used SW_SPI constructor, because it worked on it's own.
With the  HW_SPI it displays only a little stroke of pixels.
The joined image is created with SW_SPI. Then i upload the same sketch with HW_SPI and only in the middle there is a stroke of pixels moving around. I blurred lower part of the image.
[image]

olikraus, 2020-07-01:
Maybe you can now apply the setBusClock command again. Will it help? Maybe start with 100000 (100kHz) and increase that value.

alnitac, 2020-07-01:
I tried several values without any change.
Software SPI works, Hardware SPI displays nothing more than a stroke of pixels, but only if something was displayed before.
I understand that even clearBuffer() does nothing with hardware SPI.

olikraus, 2020-07-02:
Maybe you should get Hardware SPI to work reliable first. If you use one of the existing U8g2 examples, will HW SPI work?

alnitac, 2020-07-02:
This sketch works perfectly with SW SPI. 
It does not work with HW SPI : nothing on the display.
[image]

[TestOled.txt](https://github.com/olikraus/u8g2/files/4862779/TestOled.txt)

olikraus, 2020-07-02:
Summary:
 * Both devices (OLED and BMP) are connected to the same SPI bus, so the SPI pins are shared. 
 * There are two different CS lines, one for OLED and one for BMP
 * SW SPI works for the OLED
 * HW SPI does NOT workf for the OLED
 * BMP Lib will use HW SPI 

The point is: Both libs MUST use HW SPI. otherweise they will conflict with each other. If you use SW SPI with the oled, then the pins are reprogrammed to work as normal GPIO pins and the SPI hardware is deactivated. As a consequence the BMP will not work. On the other side, BMP will activate HW SPI subsystem, then u8g2 SW SPI will not work, be cause it assumes the SPI pins to be normal GPIO pins.

As a consequence: Both libs MUST use HW SPI. 

> It does not work with HW SPI : nothing on the display.

This needs to be solved. Once this is solved, the both libs will work together.

alnitac, 2020-07-04:
I tried every constructor 128X128 pixels using HW SPI : no pixel appears on display.
Sorry, my skills are insufficient to dive into the library, otherwise i am ready to help.

olikraus, 2020-07-04:
In order to help, we need more information.
Which u8g2 HW SPI constructor did you use (including args)?
Which Arduino Board do you use?
How are pins connected between Board and Display?

alnitac, 2020-07-04:
Working constructor is : 
U8G2_SH1107_PIMORONI_128X128_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 12, /* reset=*/ 8);
NOT working constructor is : 
U8G2_SH1107_PIMORONI_128X128_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=/ 10, / dc=/ 12, / reset=*/ 8);

I tried on two different boards with same results : Arduino Nano 33 IoT and Arduino Nano 33 BLE

Pin connections : 

[image]

olikraus, 2020-07-04:
Connection seems to be correct. It might still be a speed problem. You need to place the bus speed command before u8g2.begin.

alnitac, 2020-07-05:
I tried a lot of values, still nothing on display. :(

olikraus, 2020-07-07:
Hmm.. I also do not know, what is wrong here... To solve your initial problem, you could connect your display to complete different GPIO pins. I mean: Do not use D11 and D13 but something else with `U8G2_SH1107_PIMORONI_128X128_F_4W_SW_SPI`.

alnitac, 2020-07-07:
Indeed it works with software SPI, but sendBuffer() takes more than 250 ms to complete, that's too slow as i need to mesure pression at least every 100 ms.

olikraus, 2020-07-11:
not sure, what else I could suggest here...

closed 2022-03-04 by olikraus