olikraus/u8g2 #1233: Osram Pictiva 288x48 Oled Support (pixel mapped SSD0332)

DjBurba, 2020-06-30:
Hi!
I recently bought an osram Pictiva OS288048PQ33MY0C11 Oled display, 288x48 pixels, pretty nice and really cheap, but it has a particular hardware, long story short osram used an SSD0332 controller (wich seems to be an rgb controller for a 96x48 display) and built a wide monochrome display by interlacing pixels of the 3 colors. 

The bad thing about this display is that there is no documentation about it, the good thing is that because of its architecture it's sold really really cheap on ebay. It needs some level shifting (it's a 3v3 logic) and a 12v power supply (a little step up circuit is enough)...

I found an arduino library that actually works with this display and i made it work (tested some examples and some functions on an arduino uno), but it's limited to basic functions.
This is the library: https://github.com/sumotoy/OLED_pictivaWide

I'm not really good at reading and understanding libraries, but i was hoping that it would be enough to understand how to drive the display, and more importantly, if it could help adding this display to the already big family of u8g2 displays.

olikraus, 2020-07-01:
Difficult. Could be done with u8g2, but effort would be very high due to the strange memory architecture of the display. I have no time to work on this.

DjBurba, 2020-07-01:
Well... At least i tried!
i would love to know how to do it myself, but if it's difficult for you i dont think i will be able to do it...
Anyway thank you for your amazing job, almost every display i own works with this library and i dont know what i would do without it! Keep it up!

olikraus, 2020-07-02:
> i would love to know how to do it myself, but if it's difficult for you i dont think i will be able to do it...

U8g2 has been written with a specific memory architecture in mind. While this is still flexible, but if the target memory deviates too much, then the implementation will become complicated or even impossible without huge changes to u8g2.

closed 2022-03-04 by olikraus