olikraus/u8g2 #1234: Support for SSD1315Z with charge pump

liamlacey, 2020-07-02:
Hi,

I'm attempting to use u8g2 with a 128x64 OLED using the SSD1315Z driver IC and which uses a built-in charge pump.
I've read that the SSD1315 is compatible with the SSD1306, however I've tried using the 1306 device constructors but with no luck.

Looking at the datasheet I have, part of the "power on sequence with charge pump" is as follows:

> 3. After VDD become stable, wait at least 20ms (t0), set RES# pin LOW (logic low) for at least 3us (t1)(3)and then HIGH (logic high).
> 4. After set RES# pin LOW (logic low), wait for at least 3us(t2). Then input commands with below sequence:
>     a. 8Dh for enabling internal charge pump 
>     b. AFh for display ON

Is there currently a device constructor that does this specific reset pin toggling followed by these specific input commands? If not, where and how in the code can I attempt to modify an existing device constructor to do this?

Many thanks.

olikraus, 2020-07-02:
hmm... Seeedstudio claims, that the SSD1306 driver should work: https://wiki.seeedstudio.com/Grove-OLED-Display-0.96-SSD1315/

Actually the internal charge pump is enabled in the SSD1306 driver:
https://github.com/olikraus/u8g2/blob/0966f30f7daeb9249f1a4d2042460932a0b2ac48/csrc/u8x8_d_ssd1306_128x64_noname.c#L52

liamlacey, 2020-07-02:
Many thanks for the quick response - I've finally found the issue which was elsewhere - all working fine with ssd1306_128x64_noname. Sorry for the false alarm.

Thanks!

closed 2020-07-02 by liamlacey