olikraus/u8g2 #1235: Tabs and spaces

ArminJo, 2020-07-08:
Hi Oli,
thanks for this amazing library! 🥇 

Browsing through your code would be a bit easier, if you could convert all tabs to spaces.
(and remove trailing spaces).
This would be wonderful.
Thanks in advance.

Armin from Cologne

olikraus, 2020-07-11:
Does the tab size break the format somehow? Tab size should be 8 spaces.

ArminJo, 2020-07-11:
My tabs are Eclipse wide set to 4 spaces and yes, it breaks the format of your code.
You use tabs and spaces randomly, so it would be nice to be consistent.
BTW. In our software development department, tabs are banned.

olikraus, 2020-07-12:
As I said, the tab size is 8 blanks. This should be consistent (although I can not ensure this for any pull requests).
 
> You use tabs and spaces randomly, so it would be nice to be consistent.

Any specific example would be great.

ArminJo, 2020-07-12:
line 81 ff. in U8x8lib.cpp
[image]

olikraus, 2020-07-12:
Hmm... as mentioned, you should use 8 spaces per tab and not 4. So I would say your editior setting needs to be changed. Moreover this should be consistent throughout all files. Any example which demonstrates any inconsistency?

ArminJo, 2020-07-12:
> you should use 8 spaces per tab and not 4

Thank you for the valuable advice.

But tell me. why do you sometimes use spaces and sometimes tabs for indentation?

olikraus, 2020-07-12:
I do not use tabs for indentation. Instead a tab is used to replace 8 spaces. 
Indentation is 2 spaces in my code, so at indent level 4 my code editor will insert once tab.

Oli (Exil-Franke)

ArminJo, 2020-07-12:
Ok, now I understand.
This saves a few bytes on the disc, but gives strange results on editors not following the tab is 8 spaces rule. 😟

olikraus, 2020-07-12:
Actually, I assume my editor just moves to the next tab stop.

ArminJo, 2020-09-20:
So I dare there will be no solution, but it is your code, and others have to adjust.

closed 2020-09-20 by ArminJo

olikraus, 2020-09-20:
Well... Probability yes