olikraus/u8g2 #1237: SH1107 controller on a 64x128 display only show a small part of expected view

Klaus-Pedersen, 2020-07-11:
Hello,
I just received a pair of 64x128 oled displays based on SH1107 controller and I2C bus.
When I use U8G2_SH1107_64X128_2_HW_I2C as a constructor, only a very small part of the display will show something.
I guess that the display needs another initializer string (settings).
The display is attachd to Arduino Mini Pro 3.3V 8MHz.
No problems if a 32x128 display based on SSD1306 is attached to the Arduino
My hello world code look like this:
#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>
// U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0);
U8G2_SH1107_64X128_2_HW_I2C u8g2(U8G2_R1, U8X8_PIN_NONE);
void setup(void)
{
u8g2.begin();
u8g2.setFont(u8g2_font_5x7_tf);
}
void loop(void)
{
u8g2.firstPage();
u8g2.clearBuffer();
u8g2.sendBuffer();
delay(2000);
u8g2.setCursor(0, 8);
u8g2.print("Hello World! on line 8 of 64");
u8g2.drawStr(0,24,"Hello World! on line 24 of 64");
u8g2.sendBuffer();
delay(2000);
}

[image]
[image]

olikraus, 2020-07-12:
see also #1236

olikraus, 2020-07-12:
There are many other SH1107 constructors available: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#sh1107-64x128 
They may not fit from the display size, but I think it would be worth to try one of the 128x128 constructors instead.

Klaus-Pedersen, 2020-07-12:
Hi Oli,

Thx for your quick reply.
I have tried out every 128x128 Constructor - same result.
In my search for something usefull, I have tried a libary from SparkFun;  "Sparkfun oled master"
From that libary I can print somthin readable on the display.
The trick here was to flip the width and height values so Width=64 and height=128.
I need to write text on the display 128 dots wide so that libary will not work for me.
I have been looking in your source code to try to find the init commands for the SH1107 display.
In the u8x8_d_sh1107.c file i found  u8x8_d_sh1107_64x128_noname_init_seq.
I'm not sure if it is the right source file (not u8g2...) or the correct init array, but I can see that the multiplex ratio is set to 0x7F wich should be correct for the display.
Can you pin point where I can find the init array for SH1107 constructers?
Then I will dive into source code and see if I can figure out something usefull.

olikraus, 2020-07-13:
u8g2 is based on the u8x8 procedures, so yes, u8x8_d_sh1107_64x128_noname_init_seq should be the right place.

samlow-jun, 2021-03-05:
@Klaus-Pedersen Were you able to find the issue?
Looks like I found the issue with this as I had the same. Please check here
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#sh1107-64x128

You have used this LCD code
U8G2_SH1107_64X128_2_HW_I2C u8g2(U8G2_R1, U8X8_PIN_NONE);
and it shows only He on the LCD as for me on mine
I have changed it to 
U8G2_SH1107_64X128_F_HW_I2C u8g2(U8G2_R1, U8X8_PIN_NONE);
The difference looks like in models which he created there 2_HW replace to F_HW
and your code shows blinking full text.

Give it a try)

Klaus-Pedersen, 2021-03-05:
Thank you very much remember me having an issue with this module.
I will trye it out very soon.

Again Thank you

Med venlig hilsen / Best Regards

Klaus Pedersen
ACS – Advanced Computer Systems
Overdrevsvej 7
4340 Tølløse
Denmark
Phone: +45 59120100
Mobile: +45 20201611
Mail: kp@advanced.dk<mailto:kp@advanced.dk>
Skype: +45 36997888 / klaus-paraglide.dk
CVR/VAT: DK15065850

https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#sh1107-64x128

olikraus, 2022-03-04:
closing this for now...

closed 2022-03-04 by olikraus