olikraus/u8g2 #1238: Actually it looks like the last page of the OLED is not printing anything

olikraus, 2020-07-13:
Actually it looks like the last page of the OLED is not printing anything

Only 128 x 96 pixels are controllable in Full Buffer mode

`U8G2_SSD1327_EA_W128128_F_HW_I2C screen(U8G2_R0, /* reset=*/U8X8_PIN_NONE);`

_Originally posted by @NikilRagav in https://github.com/olikraus/u8g2/issues/397#issuecomment-657642667_

olikraus, 2020-07-13:
If you look into u8x8_d_ssd1327.c on your local harddisc, you should find the following lines:
https://github.com/olikraus/u8g2/blob/0966f30f7daeb9249f1a4d2042460932a0b2ac48/csrc/u8x8_d_ssd1327.c#L533-L547

Can you modify the 0x000 value in 535 (or 543) ? Will it solve the problem?

NikilRagav, 2020-07-13:
Sure let me check it out. Do you know what I'd need to change it to?
I coud look at the datasheet but if you have an idea lmk

NikilRagav, 2020-07-14:
So I switched to U8G2_SSD1327_MIDAS_128X128_F_HW_I2C screen(U8G2_R0,
 /* reset=*/U8X8_PIN_NONE);

And that constructor looks to be working correctly.

olikraus, 2020-07-26:
ah, ok :-)

NikilRagav, 2020-07-26:
Maybe it's worth removing the EA constructor or making it work the same as
the MIDAS?

Also, any tips on getting 60fps? I cranked up I2C to 1Mbps but that gets me
roughly 15-20 fps

olikraus, 2020-08-03:
I2C in general slow due to the overhead.
Also note that the SSD1327 is a 4 bit graylevel display. For each one pixel u8g2 needs to transfer 4 bit. This means the pure I2C data rate has to be devided by 4 to get the pixel rate.

closed 2022-03-04 by olikraus