olikraus/u8g2 #1239: u8g2 and RTC not working together

Asolex63, 2020-07-14:
When I use the program as below, the RTC time value gets messed up. If I don't print anything to the display the time keeps corrrect. When I use the adafruit library the RTC does not get messed up but then the display format is not correct. It is not an adafruit dispaly so that makes sense. What am I missing?

[image]

[image]

JamesGKent, 2020-07-20:
that RTC supports only 100KHz I2C clock, but i think your display supports 400KHz, U8G2 will set the bus speed to its requirements, so you might need to either set the bus speed back to 100KHz after writing to the display, or tell u8g2 to use 100KHz to the display.

Asolex63, 2020-07-21:
Thanks for helping me out, that is exactly what I missed, 
Added the statement   Wire.setClock(100000);   and it works like a charm.

closed 2020-07-21 by Asolex63

olikraus, 2020-07-26:
:-)