olikraus/u8g2 #1242: LCD Dot Matrix Screen Refreshes every loop Iternation

slickmezzie, 2020-07-19:
Hi Oliver,

Great library - thanks for sharing!

I am have a few issues displaying my ECG graph on my dot matrix (128x64) screen. The screen seems to refresh every loop iteration rather than after 127 loops (i.e. full screen)...

All I get is a line segments moving across the screen (the screen does clear every 127 iteration as the title test flickers)... Obviously I would like all the line segments to remain on the screen until I refresh the screen after it is full.

My very basic code is below (I am no coder by an stretch of the imagination!) please be gentle:

It is my code not your library but I can't see why it is not working...

Setup...

void loop(void) {
  u8g2.firstPage();
   do {
    if(x>127)  
  {
    // Clear Display
    u8g2.clear(); // Works
    x=0;
    lastx=x;
  }
    // u8g2.firstPage();
    // Display Heading text
    u8g2.setFont(u8g2_font_ncenB08_tr);
    u8g2.drawStr(35,10,"ECG Display");
    int sensorValue = analogRead(A0);
    // print out the sensorValue to serial port:
    Serial.println(sensorValue);
    // delay(1);        // delay in between reads for stability
    // Diplay scaled ECG data
    // Draw line segment
    y=(63-(sensorValue/16));
    u8g2.drawLine(lastx, lasty, x, y);
    // delay(10); test
    lastx=x;
    lasty=y;
    // Increment x 
    x++;
  }
  while (u8g2.nextPage());
}
Thanks in advance.

Regards,
Dave.

olikraus, 2020-07-26:
Did you use the `_F_` constructor?

slickmezzie, 2020-07-26:
No I didn’t, I will look into it (I do remember seeing it when I was investigating why it wasn’t working - I should have investigated it further...).

Apologies and thanks for getting back to me.

Regards,
Dave

closed 2022-03-04 by olikraus