olikraus/u8g2 #1243: Use two SSD1306 with u8g2

orel56, 2020-07-20:
Hello from France !

First, thank you for your work and your great library !

For a project I need to use two SSD1306 connected to the same arduino micro board. To my mind, my main issue come from setting the two I2C addresses of the devices, to do so I use "setI2CAddress" on each display to set de I2C address of the first one on 0x3c*2 and the other one on 0x3d*2. The fact is that both screens draw what must only be drawn on the 0x3c*2 connected screen... 

I looked for this issue on other topics but I didn't find anything.

Thank you by advance ! 

Here is my arduino code : 

```c
#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

U8G2_SSD1306_128X64_NONAME_1_HW_I2C display0(/* rotation=*/ U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
U8G2_SSD1306_128X64_NONAME_1_HW_I2C display1(/* rotation=*/ U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

void printLayout() {
  
  display1.setFont(u8g2_font_ncenB14_tr);
  display0.setFont(u8g2_font_ncenB14_tr);

  String labelLeft;
  String labelRight;

  display0.firstPage();
  do{ 
  labelLeft = "< hello";
  labelRight = "           world >";
  display0.drawStr(0,50 , labelLeft.c_str());
  display0.drawStr(0, 25, labelRight.c_str());
  } while (display0.nextPage());

  display1.firstPage();
  do{
  labelLeft = "< works";
  labelRight = "                  it >";
  display1.drawStr(0,50 , labelLeft.c_str());
  display1.drawStr(0, 25, labelRight.c_str());
  }while(display1.nextPage());
}

void setup() {
 display0.setI2CAddress(0x3c*2);
 display1.setI2CAddress(0x3d*2);
 
 display0.begin();
 display1.begin();
 
 display0.setFlipMode(0);
 display1.setFlipMode(1);
 
 display0.setPowerSave(0);
 display1.setPowerSave(0);
}``

olikraus, 2020-07-26:
hmmm... I do not see a problem here. Did you solve the issue?

orel56, 2020-07-27:
Hello, 

I may have found something, the I2C address must be change on the screen by moving a resistor nonetheless I didn't try yet ! 

I'll tell you if it works !

orel56, 2020-07-31:
Hello, it worked well !

closed 2022-03-04 by olikraus