olikraus/u8g2 #1246: I have trouble with the unifont font !

hacuongvv, 2020-07-26:
I am using 128x64 lcd, I use the unifont font to display the language of my country, the problem is that I can only display 4 lines on the lcd, is there any way to shorten the number of pixels, I want to display display 5 to 6 lines on the lcd to see more, if possible, can you help me convert the font, thank!

[https://drive.google.com/file/d/16jmLBychJ-H9ANOBVWwmAZDI_nlBkRKZ/view?usp=sharing](url)

olikraus, 2020-07-26:
1) The height of the unicode glyphs is usually 10. So it should be no problem to have 5 or 6 lines on a 128x64 display. What is the exact problem?
2)  The generation of a u8g2 requires two steps: (a) convert ttf to bdf and (b) convert bdf to the internal format. For the first step you can use fontforge or otf2bdf. The second step is described in the FAQ:
https://github.com/olikraus/u8g2/blob/0966f30f7daeb9249f1a4d2042460932a0b2ac48/doc/faq.txt#L233-L251

olikraus, 2020-07-26:
BTW: What exact language do you need. Maybe there are other fonts in u8g2 which support your language. Did you check the available u8g2 fonts?

hacuongvv, 2020-07-26:
I tried with the Vietnamese language, when writing the characters " Ắ ", the number of pixels is about 15 pixels, which is quite inconvenient: (((

closed 2020-07-27 by hacuongvv