olikraus/u8g2 #1247: UC1608 using 8bit interface ( DEM 240064D FGH-PW )

Ryder17z, 2020-07-28:
I would like to use this lib for this [display](https://www.tme.eu/en/details/dem240064dfgh-pw/lcd-graphic-displays/display-elektronik/dem-240064d-fgh-pw/) but it seems N-bit modes aren't supported.
(8bit in this case, but I'm sure you can find 4bit modules too)

[image]

Additionally, I would like some hints on bias and segment voltage since the datasheet is lacking info on that.

Ryder17z, 2020-07-29:
solved after much digging.

olikraus, 2020-08-03:
So, I assume you found this page: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp

Ryder17z, 2020-08-03:
I did, but i have trouble understanding it.

```
const int grot = 0; // i think?

const int gd7 = 53;
const int gd6 = 51;
const int gd5 = 49;
const int gd4 = 47;

const int gd3 = 43;
const int gd2 = 41;
const int gd1 = 39;
const int gd0 = 37;

const int genable = 27;
const int gcs = 31;
const int gdc = 29;
const int greset = 33;

U8G2_UC1608_DEM240064_F_6800 u8g2(grot, gd0, gd1, gd2, gd3, gd4, gd5, gd6, gd7, genable, gcs, gdc, greset);
```

> error: expected constructor, destructor, or type conversion before '(' token
 U8G2_UC1608_DEM240064_F_6800 (grot, gd0, gd1, gd2, gd3, gd4, gd5, gd6, gd7, genable, gcs, gdc, greset);

also a better datasheet for the same driver. https://www.crystalfontz.com/controllers/UltraChip/UC1608/

olikraus, 2022-03-04:
hmm... can we close this issue?

Ryder17z, 2022-03-04:
I dunno what the error was about. I ended up using a different display so I think it can be closed (i forgot about opening an issue here)

closed 2022-03-04 by Ryder17z

olikraus, 2022-03-04:
No problem, thanks for the feedback.