olikraus/u8g2 #1249: esp8622 2.23.18 vs higher version

wchpikus, 2020-07-30:
Could you check, why old version work correct (the same software) but any upper 2.23.18 make reboot?
What is the differences between?
I just update working library to new version, and not working..

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld

Soft WDT reset

olikraus, 2020-08-03:
U8g2 is tested on several official boards. It is impossible for me to test all existing baords.
To me it looks like you need to increase the watchdog timeout on you board.

olikraus, 2020-08-03:
Some ideas:
https://github.com/esp8266/Arduino/issues/34
https://techtutorialsx.com/2017/01/21/esp8266-watchdog-functions/

closed 2022-03-04 by olikraus