olikraus/u8g2 #1250: Support for Visionox M01033 SH1106G 128x32

benjaminauer, 2020-08-02:
I have tried the constructor U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI which works fine from the driving side of things.
But the line offset should be 0x10 (16 lines)
I tried to add my own constructor to apply the correct values, for more practical functionality but i couldnt really figure out, ended up getting some warnings at the end end it failed.

Datasheet for the Part: https://eu.mouser.com/datasheet/2/835/Visionox_01062017_M01033-1217761.pdf

I am not too sure if you want to add constructors for each variant.
Thanks alot

olikraus, 2020-08-16:
Thanks for pointing this out.
Did you change the code

The code for the sh1106 is here:
https://github.com/olikraus/u8g2/blob/0966f30f7daeb9249f1a4d2042460932a0b2ac48/csrc/u8x8_d_ssd1306_128x64_noname.c#L412

and refers to the ssd1306 init sequence

Did you change the code here (from 0 to 16) to crosscheck whether your display works correctly?

https://github.com/olikraus/u8g2/blob/0966f30f7daeb9249f1a4d2042460932a0b2ac48/csrc/u8x8_d_ssd1306_128x64_noname.c#L50

benjaminauer, 2020-08-26:
Have tried that. But can not really confirm anything as i struggle completly with platform.io
the changes made have no effect at all, probably some user issues right there.
quickly created my own test library where i get correct start line and offset for the display, using the recommended setup.

olikraus, 2020-08-26:
I don't know platform.io :(

benjaminauer, 2020-08-28:
I figured it out and got it working.
Created a separate file
[u8x8_d_sh1106_128x32_visionox.c.txt](https://github.com/olikraus/u8g2/files/5144323/u8x8_d_sh1106_128x32_visionox.c.txt) and added the neccesarry constructor and setup function.
The init sequence and the display info got changed compared to other sh1106 setups, and chose the tile buffer size from other 128x32 setups.

`class U8G2_SH1106_128X32_VISIONOX_F_4W_HW_SPI : public U8G2 {
public: U8G2_SH1106_128X32_VISIONOX_F_4W_HW_SPI(const u8g2_cb_t *rotation, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE) : U8G2() {
u8g2_Setup_sh1106_128x32_visionox_f(&u8g2, rotation, u8x8_byte_arduino_hw_spi, u8x8_gpio_and_delay_arduino);
    u8x8_SetPin_4Wire_HW_SPI(getU8x8(), cs, dc, reset);
  }
};`

`void u8g2_Setup_sh1106_128x32_visionox_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)
{
  uint8_t tile_buf_height;
  uint8_t *buf;
  u8g2_SetupDisplay(u8g2, u8x8_d_sh1106_128x32_visionox, u8x8_cad_001, byte_cb, gpio_and_delay_cb);
  buf = u8g2_m_16_8_f(&tile_buf_height);
  u8g2_SetupBuffer(u8g2, buf, tile_buf_height, u8g2_ll_hvline_vertical_top_lsb, rotation);
}`

Maybe you want to add it to the library. Cheers.

olikraus, 2020-09-01:
Not sure when I will have time to work on this...

olikraus, 2020-12-26:
implemented...

closed 2020-12-26 by olikraus