olikraus/u8g2 #1251: struggling with STM32F103C8T6 and ST7920

vores8, 2020-08-05:
Hi All,

I am desperately trying to make U2G@ work with STM32F103C8T6 and ST7920. I've read lots of refs like

https://github.com/olikraus/u8g2/issues/356
https://github.com/olikraus/u8g2/wiki/setup_tutorial
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform.

but still have gardage on the display.
I have the follwing setup
CLK: PA1->E
DATA:PA2->R/W
CS:PA3->RS
RESET:PA4->RST
PSB on ST7920 is connected to GND

code in main.c is
```
  /* USER CODE BEGIN WHILE */

  u8g2_Setup_st7920_s_128x64_f(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8g2_gpio_and_delay_stm32);
  u8g2_InitDisplay(&u8g2);
  u8g2_SetPowerSave(&u8g2, 0);

  HAL_Delay(100);

  u8g2_ClearDisplay(&u8g2);
  u8g2_SetFont(&u8g2, u8g2_font_5x7_tf);

  u8g2_ClearBuffer(&u8g2);
  u8g2_DrawStr(&u8g2, 0, 24, "Hello world!");
  u8g2_SendBuffer(&u8g2);

  while (1)
  {
  		u8g2_DrawLine(&u8g2, 10, 10, 100, 100);
		HAL_Delay(1000);
		u8g2_SendBuffer(&u8g2);
		HAL_Delay(1000);

    /* USER CODE END WHILE */

    /* USER CODE BEGIN 3 */
  }
  /* USER CODE END 3 */

```
u8g2_gpio_and_delay_stm32 is
```
uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr) {
	switch (msg) {
//Initialize SPI peripheral
	case U8X8_MSG_GPIO_AND_DELAY_INIT:
		delay_init();
		break;
//Function which implements a delay, arg_int contains the amount of ms
	case U8X8_MSG_DELAY_MILLI:
		HAL_Delay(arg_int);
		break;
		//Function which delays 10us
	case U8X8_MSG_DELAY_10MICRO:
		delay_us(10);
		break;
		//Function which delays 100ns
	case U8X8_MSG_DELAY_100NANO:
		__NOP();
		break;
	case U8X8_MSG_DELAY_NANO:
		break;
		//Function to define the logic level of the clockline
	case U8X8_MSG_GPIO_SPI_CLOCK:
		if (arg_int)
			HAL_GPIO_WritePin(CLOCK_LCD_GPIO_Port, CLOCK_LCD_Pin, GPIO_PIN_SET);
		else
			HAL_GPIO_WritePin(CLOCK_LCD_GPIO_Port, CLOCK_LCD_Pin, GPIO_PIN_RESET);
		break;
		//Function to define the logic level of the data line to the display
	case U8X8_MSG_GPIO_SPI_DATA:
		if (arg_int)
			HAL_GPIO_WritePin(DATA_LCD_GPIO_Port, DATA_LCD_Pin, GPIO_PIN_SET);
		else
			HAL_GPIO_WritePin(DATA_LCD_GPIO_Port, DATA_LCD_Pin, GPIO_PIN_RESET);
		break;
		// Function to define the logic level of the CS line
	case U8X8_MSG_GPIO_CS:
		if (arg_int)
			HAL_GPIO_WritePin(CS_LCD_GPIO_Port, CS_LCD_Pin, GPIO_PIN_SET);
		else
			HAL_GPIO_WritePin(CS_LCD_GPIO_Port, CS_LCD_Pin, GPIO_PIN_RESET);
		break;
		//Function to define the logic level of the Data/ Command line
	case U8X8_MSG_GPIO_DC:
		break;
		//Function to define the logic level of the RESET line
	case U8X8_MSG_GPIO_RESET:
		if (arg_int)
			HAL_GPIO_WritePin(RST_LCD_GPIO_Port, RST_LCD_Pin, GPIO_PIN_SET);
		else
			HAL_GPIO_WritePin(RST_LCD_GPIO_Port, RST_LCD_Pin, GPIO_PIN_RESET);
		break;
	default:
		return 0; //A message was received which is not implemented, return 0 to indicate an error
	}
	return 1; // command processed successfully.
}

```
There is definitely some traffic on SPI line (snapshot of th 10 sec from the above code, channels from top: CLK, DATA, CS, RESET)
[image]
Closer look at each DrawLine/SendBuffer sequence
[image]
CLK pulse shape
[image]

I've managed to run U8G2 test using the same display and Arduino, but completely stuck using STM32. 
Any help appreciated.
Thanks in advance.

vores8, 2020-08-06:
I've checked working Arduino code and it has different CLK timings
[image]

The problem is that no matter what I put into DELAY cases in u8g2_gpio_and_delay_stm32 - CLK timings do not change/ Is there something hardcoded in the library?

vores8, 2020-08-07:
nevermind. switched to HW SPI (https://github.com/olikraus/u8g2/issues/840) and it works fine

closed 2020-08-07 by vores8

klen25, 2021-07-02:
hi @vores8 , how do you solve the problem of "break" statement and change from "symbol font" to "normal font" in #840 ?