olikraus/u8g2 #1253: U8G2 blocks the main loop/task ESP32

DamithVG, 2020-08-11:
Hello.

I am using the u8g2 with ESP 32, and ssd1309 OLED 128864 display in Arduino environment. I use [Nkolban](https://github.com/nkolban/esp32-snippets/tree/master/hardware/displays/U8G2#esp32-u8g2-library-support)'s library to interface with ESP32 . Problem is, U8g2 seems to be blocking the main loop (looptask) of arduino. Updating the display is also very slow (about 2s between updates).  
For example, i usec this code in ESP32 

```
void loop(void) {
  u8g2_ClearBuffer(&u8g2);
  u8g2_SetFont(&u8g2, u8g2_font_timR14_tf);
  char buf[100]="";
  String(millis()).toCharArray(buf, 100);
  u8g2_DrawStr(&u8g2, 2, 17,buf);
  u8g2_SendBuffer(&u8g2);
}
```

As you can see i didn't  added any delay, but display only updates about once every two seconds. 
  
I Examined the [nkolban](https://github.com/nkolban/esp32-snippets/tree/master/hardware/displays/U8G2#esp32-u8g2-library-support) library and i saw there is a delay function added to the library. It is blocking whereever its been used. 
```
		case U8X8_MSG_DELAY_MILLI:
			vTaskDelay(arg_int/portTICK_PERIOD_MS);
			break;

```
Also i saw in the description of U8g2 library, few milliseconds of such a delay required to operate the library   `U8X8_MSG_DELAY_MILLI.` 
How can i overcome this blocking behavior of the U8g2, without creating a separate task for display purposes. 
How can i minimize this delay. 

Thank you

olikraus, 2020-08-16:
Actually I do not know anything about Neil's implementation. If you use a hardware accelerated communication (I2C or SPI subsystem of your uC) then the delays are usually not used. Usually means: During startup of a display, some commands sent to the display require little more time on the display side to get executed. In such a case, u8g2 will just do a busy delay of some milliseconds. During normal operation however this should not be the case and you should be able to reach >10 FPS with i2c.

olikraus, 2020-08-16:
Looking closer to Neil's code:

https://github.com/nkolban/esp32-snippets/blob/fe3d318acddf87c6918944f24e8b899d63c816dd/hardware/displays/U8G2/u8g2_esp32_hal.c#L67

10kHz seems to be very slow. 100kHz is supported by all of them, many can do 200kHz and some are also able to do fast I2C with 400kHz.

DamithVG, 2020-08-19:
> Looking closer to Neil's code:
>
> https://github.com/nkolban/esp32-snippets/blob/fe3d318acddf87c6918944f24e8b899d63c816dd/hardware/displays/U8G2/u8g2_esp32_hal.c#L67
>
> 10kHz seems to be very slow. [...]

what about SPI. I used SPI.

DamithVG, 2020-08-19:
Thank you. is 10fps true for SPI also?.

DamithVG, 2020-08-19:
I raised an issue [here](https://github.com/nkolban/esp32-snippets/issues/1005)

olikraus, 2020-09-01:
> Thank you. is 10fps true for SPI also?.

Maybe. It depends on the complexity of your code.

closed 2022-03-04 by olikraus