olikraus/u8g2 #1254: RLE limit reached

3s1d, 2020-08-15:
Hi,
I am trying to generate a large font, but bdfconv keeps reporting the following error:
`RLE Compress: Error, glyph too large, encoding=35 cnt=350`

As I understand the code, there is a run length limit of 255. Is there a simple way around it? I mean could a 16bit variable be used or simply placed two RLE blocks in series?

Unfortunately I can't find any docu about this part. Could you please point me into the correct direction and I'll eventually figure it out.

Thanks
Juergen

olikraus, 2020-08-16:
Yes, the maximum size of the glyph after RLE encoding is only 255 bytes. And yes, the corresponding entry in the internal format is one byte only.

Dokumentation is here: https://github.com/olikraus/u8g2/wiki/u8g2fontformat, see the jump offset to the next glyph.

For a glyph with encoding >= 256, the offset is used here: 
https://github.com/olikraus/u8g2/blob/0966f30f7daeb9249f1a4d2042460932a0b2ac48/csrc/u8g2_font.c#L706

In bdfconv the size update happens here: 
https://github.com/olikraus/u8g2/blob/0966f30f7daeb9249f1a4d2042460932a0b2ac48/tools/font/bdfconv/bdf_rle.c#L504-L511

So I assume, these two locations need to be updated for 16 bit.

On the other side: Is it really worth this effort. You could split the glyph into two pieces maybe. In fact I observed, that glyph size up to 100 pixel seem to be possible with the RLE encoding. Do you want to implement a bitmap? If so, then also splittig into multiple smaller pieces could be an option.

olikraus, 2020-08-16:
Oh, some update might be required here also:
https://github.com/olikraus/u8g2/blob/0966f30f7daeb9249f1a4d2042460932a0b2ac48/tools/font/bdfconv/bdf_rle.c#L394-L396

closed 2020-09-01 by olikraus

3s1d, 2020-09-01:
Thanks. With your help, I figured it out :)