olikraus/u8g2 #1256: using custom HW I2C bus

wanthalf, 2020-08-16:
Hi, is there any possibility to enforce the use of a custom HW I2C bus other than inserting `#define Wire Wire1` directly into `src/U8x8lib.h`? I tried several other #defines used by the library in my main sketch, but nothing worked.

olikraus, 2020-09-01:
Wire1 should be supported by the constructors with 2ND_HW_I2C at the end, see here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp

wanthalf, 2020-09-02:
Oh, sorry, somehow I overlooked this.
Meanwhile, I also found out in the discussions, how to set an alternative I2C address and why 0x78 corresponds to 0x3C and 0x7A to 0x3D. It would also be worth describing explicitly in the setup documentation.
Thanks.

olikraus, 2020-09-03:
It is described in the user manual.

closed 2022-03-04 by olikraus