olikraus/u8g2 #1258: 0.96 inch OLED display 64×128 LCD module SSD1107

mclareniak, 2020-08-20:
Could you add a display SSD1107 64x128 I2C?
https://www.aliexpress.com/item/4000464960639.html?spm=a2g0s.9042311.0.0.27424c4dvcpGI0
Thank you.

olikraus, 2020-08-20:
It should be there:
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#sh1107-64x128

U8G2_SH1107_64X128_1_SW_I2C
or
U8G2_SH1107_64X128_1_HW_I2C

mclareniak, 2020-08-20:
It's probably not there. I had to add the row manually because it wasn't there. When I added it manually, an error occurred while compiling the code.
Check image.
`https://ibb.co/3y0tnVx`

olikraus, 2020-08-20:
The mentioned constructor existis here: https://github.com/olikraus/u8g2/blob/master/cppsrc/U8g2lib.h
Did you use the latest version?
You could also jump to the lastest (stable) beta:

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

closed 2020-09-01 by olikraus