olikraus/u8g2 #1259: Question: including fonts/font libraries

marcdraco, 2020-08-20:
Still very new to this - but I have your test code working beautifully across three different clone Arduino boards and two different displays. So, pretty happy with that!

My next job is to incorporate a large (32pix height) display font (Iconic) so my system doesn't have to spit text out - pictures are universal after all. But I am unable to follow what I'm supposed to do. The script you've provided appears to be designed for your machine and although I could hack that, I'm sure there must be a better way. My main dev system is on Windows (I know, I know, but it's what I got) and the Linux box is rather modest (and even that's a stretch).

Are you able to point me in the right direction? I could always just use my own binary and hard code it but Iconic seems a much better option as I'm sure you'd agree.

Amazing work on the graphics library though. My Seeeduino Xiao board just hangs when Limor's code attempts to set 3.3v during the initialisation phase and although your constructor system took me a while to wrap my aging and lockdown-fogged head around, once I did have the hang of it, I was off and away like a greyhound after a rabbit!

olikraus, 2020-08-20:
Nicely written request. Nevertheless a url to the mentioned iconic font would be great.
U8g2 already includes open iconic font in a 32x32 pixel verions. Is this what you need?

[image]

Details are here:
https://github.com/olikraus/u8g2/wiki/fntgrpiconic

In general it is a good idea to look around the existing fonts and icons shipped with u8g2:
https://github.com/olikraus/u8g2/wiki/fntgrp

marcdraco, 2020-08-20:
Yes! That's amazing and just perfect Oliver, thank you.

I had no idea! I'm just plodding through the library functions and trying them out - that's some piece of work you've done there and the speed of this thing! I'd always (*wrongly) assumed that things like the Mega would have quite modest performance, particularly over a serial I2C line - but woah, **_was I wrong_**!

There's way more than I need there so now I'm going back to the "beer mat" to figure out how best to make use of these. I can even imagine making something completely pointless (but amusing) by plotting an audio waveform in real-time. A bunch of characters like this shaves hours or even days of tedious work though. I gotta get my head around your Wiki though, because I must have missed this.

One of the annoying things about getting older is I forget stuff - but I'm sure I had your library working (doing some sort of demo) on the Seeeduino Xiao which as I've said isn't possible on Limor Freid's otherwise excellent work. Not her fault of course, there are so many of these controllers with their own quirks.

This particular project is for my daughter's boyfriend, lovely lad but 17 and with the attention span of a gnat. Assuming I don't bludgeon him to death with it, I'll drop you a small contribution for your work with my greatful thanks. Won't reflect the amount of work you've put in of course, but (always assuming it's not a rat's nest) I'll open up my own code for others to use.

closed 2020-08-20 by marcdraco

olikraus, 2020-08-20:
I would say this: we all become more experienced over time. Even noticed that I started to forget things. 😄

marcdraco, 2020-09-26:
All good Oliver. We've had to take a breather to do something else (ugh) but related and the result is I've found myself switching to Visual Studio with Platform IO because it was getting far too unwieldy for the Arduino editor.

Still, when I'm not pulling out my "air hair" (that's hair for bald dudes to pull out) I'm having a great time learning new stuff!