olikraus/u8g2 #1263: Convert from openGLCD

crd5014, 2020-08-24:
I am converting a project from openGLCD to U8G2. Two questions:
1) Is there a way to convert an openGLCD formatted custom font library to a font library compatible with U8G2?  Maybe several steps, convert openGLCD (i.e. GLCDFONTDECL hex data) to BDF then to U8G2?
2) Is there method or function similar to the openGLCD gText that allows you to define differently formatted text areas on a given GLCD screen and print to each like a terminal?

Thank you very much for any advice you can provide.

olikraus, 2020-09-01:
> Is there a way to convert an openGLCD formatted custom font library to a font library compatible with U8G2? Maybe several steps, convert openGLCD (i.e. GLCDFONTDECL hex data) to BDF then to U8G2?

No. Maybe the source of that font is available as ttf or bdf, then it could be converted.

> Is there method or function similar to the openGLCD gText that allows you to define differently formatted text areas on a given GLCD screen and print to each like a terminal?

Not directly. There are some helper function like https://github.com/olikraus/u8g2/wiki/u8g2reference#setclipwindow

Due to the nature of u8g2 there is no bit blit (https://en.wikipedia.org/wiki/Bit_blit) which is there in GLCD to do the sub-window scrolling. Still such kind of effects are possible in u8g2 (see the scrolling examples) but they are based on redrawing and not on bit blit procedures like in GLCD. So as a result: There are no simple 1:1 commands available.

You may ask: Why is there such a difference? The answer is simple: U8g2 has been designed to work without reading back the memory buffer. In fact many some displays do not allow reading back the graphics display content, so such displays are supported by u8g2 which will be never supported by GLCD.

crd5014, 2020-09-01:
OK, thanks for the insight.

dpwe, 2022-02-21:
I was able to convert some openGLCD fonts (the calfonts) to BDF, and then use them within u8g2.

See the IPython notebook here: https://github.com/dpwe/calfonts/blob/add-bd/glcd-fontcreator-to-bdf.ipynb

You should be able to use this for any of the openGLCD fonts.

olikraus, 2022-02-22:
Thanks a lot. If the bdf's are available somewhere, then I also could add them to u8g2 permanently.

dpwe, 2022-02-22:
That sounds great!  The BDF files for the two [CalFonts](https://github.com/johnmccombs/calfonts) are here:
https://raw.githubusercontent.com/dpwe/calfonts/add-bd/CalBlk36.bdf
https://raw.githubusercontent.com/dpwe/calfonts/add-bd/CalLite24.bdf
I can also make a PR for u8g2 if that's easier.

olikraus, 2022-02-22:
> I can also make a PR for u8g2 if that's easier.

there is a script which does most of the work, so I will do this, no problem.

dpwe, 2022-02-22:
Sounds great.  It would be nice to preserve the copyright message from the cal fonts files, viz:

Copyright (C) 2010 by Integrated Mapping Ltd

olikraus, 2022-02-22:
Sure, indeed I will mention (as usual) the copyright:

```
 *
 * Copyright (C) 2010 by Integrated Mapping Ltd
 * 
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 * 
 * The above copyright notice and this permission notice shall be included in
 * all copies or substantial portions of the Software.
 * 
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
 * THE SOFTWARE.
```

dpwe, 2022-02-22:
Great! Thank you.

olikraus, 2022-02-26:
32-127 (MM_R)

closed 2022-03-04 by olikraus