olikraus/u8g2 #1264: LCD COG ST7567 imcompatibilty

MpinhoCODE, 2020-08-25:
Dear,

I would like to know if you could help me with this display to connect with U8g2  library please, I tried but nothing happens.
Thank you very much for your attention. 

Best regards,
Matheus.
[HGO1286427.pdf](https://github.com/olikraus/u8g2/files/5125166/HGO1286427.pdf)

olikraus, 2020-08-25:
At least something should be visible. Maybe adjust the contrast. Is your setup correct? Maybe also post a schematic of your setup.

MpinhoCODE, 2020-08-25:
How to adjust the contrast? My setup is :

1 PSB to GND
2 C86 to GND
3 VG to 0,1F and the other end of the cap to GND
4 Xv0 to one end of another cap 0,1F
5 V0 to other end of 0,1F cap
6 Vss to GND
7 Vdd to 3V3
8 D7 (SDA) to SPI Clock
9 D6 (SCL) to SPI Mosi
10 D5 to Vdd
11 D4 to Vdd
12 D3 to Vdd
13 D2 to Vdd
14 D1 to Vdd
15 D0 to Vdd
16 /RD to Vdd
17 /RW to Vdd
18 A0 (D/C) to pin 9
19 RSTB  to pin 8
20 CSB to pin 10

MpinhoCODE, 2020-08-25:
[image]

olikraus, 2020-08-25:
Did you use a 3.3v Arduino? Use the setContrast u8g2 command or the corresponding example to cycle through all contrast values.

MpinhoCODE, 2020-08-25:
Yes, Arduino Due.

olikraus, 2020-08-25:
Then please execute the contrast test: https://github.com/olikraus/u8g2/tree/master/sys/arduino/u8g2_page_buffer/ContrastTest
It should be part of the installed examples in the Arduino IDE.
Will you see somehting?
Of not, then I assume there is a wiring problem.

MpinhoCODE, 2020-08-26:
I tried with the ContrastTest example, but I don't know which of these I can use : 
`//U8G2_ST7567_JLX12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 7, /* dc=*/ 9, /* reset=*/ 8);  
//U8G2_ST7567_JLX12864_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 7, /* dc=*/ 9, /* reset=*/ 8);  
//U8G2_ST7567_OS12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 7, /* dc=*/ 9, /* reset=*/ 8);  
//U8G2_ST7567_OS12864_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 7, /* dc=*/ 9, /* reset=*/ 8);  
//U8G2_ST7567_ENH_DG128064_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); 
//U8G2_ST7567_ENH_DG128064_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); 
//U8G2_ST7567_ENH_DG128064I_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); 
//U8G2_ST7567_ENH_DG128064I_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); 
`
It doesn't work yet...

MpinhoCODE, 2020-08-26:
> I tried with the ContrastTest example, but I don't know which of these I can use :
> 
> `//U8G2_ST7567_JLX12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=_/ 13, /_ data=_/ 11, /_ cs=_/ 7, /_ dc=_/ 9, /_ reset=*/ 8);
> 
> //U8G2_ST7567_JLX12864_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=_/ 7, /_ dc=_/ 9, /_ reset=*/ 8);
> 
> //U8G2_ST7567_OS12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=_/ 13, /_ data=_/ 11, /_ cs=_/ 7, /_ dc=_/ 9, /_ reset=*/ 8);
> 
> //U8G2_ST7567_OS12864_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=_/ 7, /_ dc=_/ 9, /_ reset=*/ 8);
> 
> //U8G2_ST7567_ENH_DG128064_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=_/ 13, /_ data=_/ 11, /_ cs=_/ 10, /_ dc=_/ 9, /_ reset=*/ 8);
> 
> //U8G2_ST7567_ENH_DG128064_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=_/ 10, /_ dc=_/ 9, /_ reset=*/ 8);
> 
> //U8G2_ST7567_ENH_DG128064I_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=_/ 13, /_ data=_/ 11, /_ cs=_/ 10, /_ dc=_/ 9, /_ reset=*/ 8);
> 
> //U8G2_ST7567_ENH_DG128064I_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=_/ 10, /_ dc=_/ 9, /_ reset=*/ 8);
> `
> It doesn't work yet... if it is a wiring problem could you help me to solve it? (If you prefer to use e-mail maybe would be faster)

olikraus, 2020-08-26:
Probably you need to try several constructor. But you need to set the correct Arduino pins. This issue should be good enough. Emails I try to avoid.

MpinhoCODE, 2020-08-26:
So the problem is the wiring between the Arduino and the display? I tried the constructors that I mentioned and it displays nothing. :(

olikraus, 2020-08-26:
did you also execute the constrast test example? Still nothing visible?
Which arguments did you use for the constructors?
According to your initial mail it should be something like:
U8G2_ST7567_JLX12864_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=/ 10, / dc=/ 9, / reset=*/ 8);

But you did not mention where you have connected the SPI pins. If you want to use HW pins, I assume it has to go to the 6-pin male pinheader in the middle.

MpinhoCODE, 2020-08-26:
Nothing visible yet. Yes I used the constructors like you mentioned. 
And I connected the SPI pins to the 6-pin make pinheader in the middle of the Due board.

MpinhoCODE, 2020-08-26:
The appearance is not so good as you can see...
[image]

MpinhoCODE, 2020-08-26:
I know the appearance is not so good.. :/
[image]

olikraus, 2020-08-26:
a prototype board is a little bit unreliable. Also the caps need to be very close to the display.

MpinhoCODE, 2020-08-26:
[image]

Is it better ?

MpinhoCODE, 2020-08-26:
Is it better this image ? 
[image]

I will try to solder the cap.

olikraus, 2020-08-26:
Best is to solder everything. Keep the wires as short as possible. Add another cap close to the display power supply lines (0.1uF)

MpinhoCODE, 2020-08-26:
> Best is to solder everything. Keep the wires as short as possible. Add another cap close to the display power supply lines (0.1uF)

As soon as I make these changes I reply here.
And I have to say that I'm very grateful for your support.

MpinhoCODE, 2020-08-26:
As soon as I make these changes I reply here.
And I have to say that I'm very grateful for your support.

olikraus, 2020-08-26:
One more comment: which values do your caps have. Your schematic is for sure wrong: 0.1F is too huge. I think 0.1uF = 0.0000001 F should be correct. However your capacitors look more like 0.1 nF = 0.0001 uF = 0.0000000001 F. I mean you really need at least 0.1 uF

MpinhoCODE, 2020-08-27:
The capacitor which I'm using is 100nF, is it the correct?
[image]

olikraus, 2020-08-27:
Yes, should be correct.

MpinhoCODE, 2020-08-27:
[image]

That's the current circuit. Still shows nothing... I used cap 1uF and 100nF. I think this display is not working, or burned... I don't know why is not working .

olikraus, 2020-09-01:
I would personally prefer a ceramic caps in both cases, but well... Difficult. I still guess there is a wiring problem. In such cases, I would take my scope to see whether the signals at least arrive at the display pins.

MpinhoCODE, 2020-09-02:
I will try to buy an fpc adpater board compatible to work with this display better. 
And my apologies but I don't know how to have 100% of sure that this library will work with this display.  
Thank you very much for all the help.

closed 2022-03-04 by olikraus