olikraus/u8g2 #1265: can not get bdfconv to work

s54b32julian, 2020-08-27:
I cant get bdf converter get to work. maybe someone can give me the C code please or tell me why bdfconv_2_22.exe says cant not open bdf CARSYMBOL.bdf
[Documents.zip](https://github.com/olikraus/u8g2/files/5138312/Documents.zip)

olikraus, 2020-08-27:
for me it works without problems.

```
./bdfconv -v -f 1 -	./bdfconv -v -f 1 -m '$$20-$$ffff' CARSYMBOL.bdf -o u8g2_font_carsymbol.c  -n u8g2_font_carsymbol -d ../bdf/helvB18.bdf 
```

[image]

``` C++
/*
  Fontname: CARSYMBOL
  Copyright: 
  Glyphs: 224/256
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_carsymbol[3819] U8G2_FONT_SECTION("u8g2_font_carsymbol") = 
  "\340\0\4\3\5\5\4\6\6\26\25\0\377\23\0\20\0\2\25\5)\16\316 \6\0 \30\2!\11\202"
  "%Y\302\3\42\1\42\11\205${\202\4\27\2#\35\212!\251\216\210\220\210\220\210\310\201\210\42\21!"
  "\21!\21\221\3\21E\42B\42\62\0$\27\310\345\250\216\24\215\205\12\12!*F\22\62\22\24*l"
  "\250d\0%\37\215%\371\6\221\210\66\42jD$\204D\310(D%\310H\204$DdD\324\210h"
  "D\1&\33\213%\311\12\231\210\224\210\224\210\30\335\234\204\224\214\10\21\205\224\214\14M\211\0'\10\202"
  "$K\302A\0(\17\4&h\212\204\210\12\21\375FD\215\0)\20\4&h\202\214\210\32\21\375B"
  "D\205\10\0*\13\246\340z\212\210\5\11\13\0+\15\350\244\251\216\230\232\203\30\61\65\0,\11\203d"
  "X\206\12\12\0-\7% jB\1.\7\42$Y\202\0/\16\205!Y\216\22\31Md\64\221\221"
  "\1\60\21\210%\251\12\215\210\26B|$!\242\15\11\0\61\11\204)\251\212\211\376\3\62\17\210%\251"
  "\12\215\210\204\220\230*}v\20\63\25\210%\251\12\215\210\204\220\230\224\320\234\62\42\11\21\31\22\0\64"
  "\25\210%\251\226\324\42\32\11\31\11\21\25\62\22\7Qb*\0\65\24\210%\251\302\301\230\62\223\21\71"
  "\61\61\42\11\21\31\22\0\66\25\210%\251\12\215\210\204\20\231\311\210\204\20G\22\42\62$\0\67\20\210"
  "%\251\302A\230*\61Ub\252\304\324\0\70\30\210%\251\12\215\210\204\20\221\204\210\14\215\210\204\20#"
  "\11\21\31\22\0\71\25\210%\251\12\215\210\204\20G\22\42#fD\22\42\62$\0:\10\2%Y\202"
  "\260\0;\14ceX\206<\220\204\4\5\0<\12\350\244\251\326\314\42\271\15=\13\210$\252\302A<"
  "\304A\0>\14\350\244\251\302\334:\241\231U\0\77\21\210%\251\12\215\210\204\220\230*\315\344\341d\0"
  "@,\357e\30\227\335\320\220\240\210L\205\210\304\210\10\211\214\10\205\220\10\205\214\14\205\214\14\205\214\210"
  "\204\222\211\21\211\203 \361)\271#\0A\67v\42hOY\225\4\21\205\320\33\21\11\13\211\211\203\7"
  "\364\20\366\20\366\20\366\20\266#\26E\61\21\26E\61\21\266#\366\20\366\20\366\20\366\20\7\17(\16"
  "\16,\0B<\265\346g\347<\304\361\1]E+\222\21\272\221e\26\23\206\7r\63\67kjF\16"
  "&\16fT\310H\311\334\214\310\34\210\310M\254+\31\251\42\31\241*\31\251\223\250\20\276\207\230\4C"
  "\62\65fh\323\201<\300<\300\201\260\224(\325\134\204\234Q\204\254\204D\204\354\301\354\301<\300\204<"
  "\300\204<\300\204<\200\210\254\15\241\325\234\350\1\21\0D\66\265\346g\347<\314<\214=\200=\300<"
  "\314<\214=\300<\314<\214=\300<\314<H\251\314\304\214H\5\213\232\203\203\270U\63j&*X"
  "\324\34\34\304\215,\2E\64u\346g\303<H\204\204\245\204\310`\305A\205\211`\205\214\364\214\360\220"
  "\354\224\254\230\250\321\301\1=\204z\10\365\0\62\342\62\342\23\363\0\365 \363@\61\0F\63\253\372\27"
  "\223`\225\210TPT\204DT\4U\204DT\4U\204DT\4U\204DT\204\32\11\211\25\42%"
  "#\24$$#\64!\23R\42\24S\65\0G\61U&hk<\320<\10=\210\204<\300\304\270\214"
  "\264H\210\354\304\204\254\310\304\344\310B\231\31\271\241\240\61Y)\251\241\231\251)\31y\0\211\203\77H"
  "\15\211)\331\202\24\257\16\254\370J\0I\10\202)i\302\7\3J\15\207!\231\226~E\303F\242\2"
  "\0K\31\211)\271\202\220\204\214\210\32\11!\252\61*\11!\21\31%B\22R\2L\13\210)\251\202"
  "\230\376\263\203\0M\25\212)\351\202\30Y\221\321\311A\210\5ODHD\310\310\4N\23\210)\311\202"
  "PM\215\211\11\205\4\205\4\211\233F\2O\24\213%\331N\321\314\210\224\204\34\337IH\211\314\14\325"
  "\0P\20\211)\271\302\211\220\204\24G\22'r\272\3Q\25\253\345\330N\321\314\210\224\204\34\337\210H"
  "\310\220\314\14\235\12R\34\212)\331\302\215\220\210\224\204\224\204\224\204\220\310\215\220\210\224\204\224\204\224\204"
  "\230\0S\24\212%\311\212\215\220\204\30\245\244\245B\62\11!\31\23\0T\14\212!\251\302\301\220\240\376"
  "#\0U\16\211)\331\202\24\377\225\204\214L\11\0V\30\214!\311\202 \241\204\230\210\230\214\220V\42"
  "b\42r\204\224\242R\0W \220!\11\203\60\225\12!%BJdh\224\220\10\211\220H\221\220\221"
  "\320\11\11\12\11\12\11\1X\32\214!\311\202\240\204\230\214\220\224\210\34\245\250$\235\210\224\220\214\230\204"
  "\240\0Y\22\214!\311\202\240\204\230\214\220\224\210\34\245\250\276\2Z\20\211%\271\302\201\234\230:\61\355"
  "\304\324\35\10[\12\4&X\202\211\376\237\20\134\16\205!Y\202\214\220\214F\62\32\311\10]\12\4\42"
  "X\2\211\376\237\30^\12fd\213\212\14\205\210\0_\10*`\250\302\301\0`\10d\244k\202\214\2"
  "a\23)%\251\206\211\220\234\222\13!\11!\11!\221\203\0b\21\210%\251\202\230\62\223\21\11!\36"
  "\225H\230\0c\21(%\231\12\215\210\204\20\231F\22\42\62$\0d\17\210%\251\232&\26\42E<"
  "\222\20\31\61e\23(%\251\12\215\210\204\20\321A\231\220\204\210\14\11\0f\15\206!i\316\210\220\210"
  "\211\220~\2g\22\250%\250\212\205H\21\217$DF\314T\211\224\0h\16\210%\251\202\230\62\223\21"
  "\11!~$i\12\202%I\202\320\301\1\1j\14\3\42H\206\234\204\376\13\12\0k\25\210%\231\202"
  "\230\66\22\42*d\210\210$dD\324H\10\11l\10\202%I\302\7\3m\27,%\351B\311\12\12"
  "\65\62\64\62\64\62\64\62\64\62\64\62\64\12n\14(%\251\202\311\210\204\20\77\22o\20(%\251\12"
  "\215\210\204\20\217$DdH\0p\21\250%\250\202\311\210\204\20\217J$L\304\64\3q\17\250%\250"
  "\212\205H\21\217$DF\314tr\13%%i\302A\210\214\276\1s\20'%\231F\205\14\225X\231"
  "\24\215D\5\0t\15\206!i\212\220\22\23!}%\1u\14(%\251\202\20\77\222\20\31\61v\20"
  "(!\211\202\20\221\204\210\66DTbj\0w\27,!\311\202\214\14\215\14\215\12\11\12\21\11\12\21"
  "\66#C:\1x\21(!\211\202\220\204\210\66TR\64\42Z\10\11y\23\250!\210\202\20\221\204\210"
[... cut ...]
  "%\251\206\210,\215\210\204\20\217$DdH\0\367\13\250\344\251\216\354A\254\14\0\370\23(%\251\212"
  "\205\210\204\210\13\11\23\23\11\21\11\23\0\371\16\250%\251\206\234Z!~$!\62b\372\16\250%\251"
  "\226\266B\374HBd\304\0\373\20\250%\251\216\24\215\210\244\20\77\222\20\31\61\374\16h%\251\206\210"
  "\244\20\77\222\20\31\61\375\24(\42\210\226\266BD\22\42\332\20Q\211\251\22S\64\5\376\23\10&\250"
  "\202\230\62\223\21\11!\36\225H\230\210i\6\377\25\350!\210\212\204\244\20\221\204\210\66DTb\252\304"
  "\24M\1\0\0\0\4\377\377\0";

```

s54b32julian, 2020-08-27:
thank you for the fast response, oli! 
I´m on Win7 x64, "CARSYMBOL.bdf" is in the same folder.
I can not open the .exe, it´s closing it self after 1 second. And if i open via command line an paste the following line in: bdfconv -f 1 -m '32-255' -n fontname -o myfont.c CARSYMBOL.bdf; (name and outputname is not important for me, only want the C-code)
it says´s "can not open bdf CARSYMBOL.bdf"

Any ideas, how can i help myself in the future? I need to optimize the custom symbol font a few times, before its perfect.

olikraus, 2020-08-28:
maybe use lowercase everywhere. I have used linux for the conversion above.

lcollins1989, 2020-08-28:
hello,

Sorry to jump in but I'm having the same problem. I've created my BDF file using fony and I've downloaded both bdfconv.exe and bdfconv2.22.exe and can't get either to work. I'm using windows 10.

bdfconv.exe doesn't work at all, I just get an error message. the other one opens and closes in 1 second. 

Any idea how I can solve this?

olikraus, 2020-08-30:
There was an issue with single/double quotes on windows. Did you check the u8g2 FAQ?

lcollins1989, 2020-08-30:
Hello,

Yes I used double quotes and followed the FAQ.

I actually opened my own thread as I felt bad hijacking this one.

olikraus, 2020-08-30:
I just tested everythig on my system and it works just fine.
I placed my test into a batch file, which is here:

https://github.com/olikraus/u8g2/blob/master/tools/font/bdfconv/test_helvb18.bat

Please check this batch file. It should work without problems.

closed 2022-03-04 by olikraus