olikraus/u8g2 #1266: System5x7 font

crd5014, 2020-08-30:
I am trying to find a U8G2 font that matches the old GLCD System5x7 font.  Is there one available?  Thanks.

olikraus, 2020-08-30:
Maybe this?
https://github.com/olikraus/u8g2/wiki/fntgrpx11#5x7

crd5014, 2020-08-30:
OK, thanks. In looking into this further, it looks like the System5x7 font may be better defined as a 6x8 font, mono-
spaced including one pixel trailing space and a line space. Attached is a screenshot of the GLCD version.
[image]

olikraus, 2020-08-30:
Maybe the 6x12 is more close:
https://github.com/olikraus/u8g2/wiki/fntgrpx11#6x12

But I do not think you will find something identical.

crd5014, 2020-08-30:
OK, thanks

closed 2020-08-30 by crd5014