olikraus/u8g2 #1267: Cannot figure out BDFconv.exe in cmd

lcollins1989, 2020-08-30:
Hello!

I finally figured out that BDFconv.exe needs to be opened with windows command prompt but I can't seem to get it to convert properly! Each time I just get the error that my bdf file cannot be opened.

I have attached it to this message, are you able to convert it for me or walk me through it.

I'm opening cmd and typing A:\Desktop\bdfconv -f 1 -m "32-255" -n aurebesh -o aurebesh.c aurebesh.bdf

Thank you and sorry to be a pain. I've been trying for hours!

[Aurebesh.zip](https://github.com/olikraus/u8g2/files/5146204/Aurebesh.zip)

olikraus, 2020-09-01:
I have used this on my linux machine (i do not have access to Windows at the moment):
```
	./bdfconv -v -f 1 -m '$$20-$$ff' Aurebesh.bdf -o u8g2_font_aurebesh.c  -n u8g2_font_aurebesh -d ../bdf/helvB18.bdf 
```
This is the output:
[image]

If I shell comment this: It looks like the lower part of the glyphs are cut off. I mean, it might be intended, but if you look at the zero, then the BDF code looks like this:
```
STARTCHAR 048
ENCODING 48
SWIDTH 1080 0
DWIDTH 15 0
BBX 14 12 0 2
BITMAP
7FF8
7FF8
C00C
C00C
C00C
C00C
C30C
C30C
C00C
C00C
C00C
C00C
ENDCHAR
```
I would have expected the 
```
7FF8
7FF8
```
before ENDCHAR, so it looks like, that something went wrong during the bdf export.
Code: 
``` C++
/*
  Fontname: Aurebesh
  Copyright: 
  Glyphs: 224/256
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_aurebesh[3330] U8G2_FONT_SECTION("u8g2_font_aurebesh") = 
  "\340\0\4\3\5\4\3\5\6\23\17\0\0\16\2\16\2\1\371\4~\14\345 \5\0\10W!\16e)"
  "MJ\62\261D\223\211%A\0\42\17\350xS\220\251\221\20Q!C\64\65\5#\31m)]\222\20"
  "\224\20\224\20\262\70\250\260\207\265\70\250\60\222\20\224\20\2$\30\215)]\21\11\24\11;xP\42\22"
  "\63\42\22#*:\252\351\24\0%&\217)\341\250\220;\10\223!\223!\223\220\70\221\220\70\210\220)"
  "\251!\222\70\210Pq\20\241,\42H,BF\0&&\217)a\220;\20\263\220\252\221\222\220\31\222"
  "\20\222\231\20\222\21\21\222\21\221\222\30\221\222\220\221\222\220\21\243\21\1'\10\344\210K \321\7(\14"
  "\204)K\20\321\223\203\21M\0)\12\204)K\321\223\203\21\35*\20\7i\323\210\230\270\250\70\270\250"
  "\270\210\230\0+\16j)Y\22\324\321\301\201\221\240F\0,\7\202(G\70\10-\7JhYx`"
  ".\11\206(O\20\341D\0/\14\204)K\321\223\20\25\42\232\0\60\22\216)\337\70(\71\250\20\345"
  "\221\20\221\20)\247\2\61\10g)]\70\313\177\62\23MI\335\70 \71\240\220\244\224\325\366\340\342\240"
  "\2\0\63\24m)\335\70\240\230\253\244\224= \71 \221\225\245\244\24\64\13\215)]\220\344\313\203o"
  "u\65\22\215)]\370\240\36\366\240\342\240XV-%\245\0\66\20\215)]\370\240\36\370\200\344\200B"
  "\222/\5\67\11m)]\70\260\325\177\70\21\216)\337\70\250\70\70\20\345\364\340\3RN\5\71\22M"
  "I\335\70 \71\240\220\344K\211\3\222\3\12\0:\11\246(Q\30\32\222R;\10\202)G\370`\0"
  "<\17J)\331\14\243\61)\232\33\254\263\243\14=\13\312HYx`\17up`>\20J)Y\210"
  "$\254+\234\33\252\251!\23\4\77\13\205)M\240\270\240\220\321\37@\33\215)\335\70\250\70\240\20\33"
  "\21\223\21)\22\241\222UB%R$&#\66\2A\32\220)e\230\225\30\25\231\224\221\24:\250:"
  "\240Gw@tP#)\62)B\31\220)\345\71\230:\250\31\234\30\235Gu\60v\60\217jtb"
  "p\2\0C\22\211)\327\323\35\211\4\211\4\211\4\211\4\235:\0D\23\215)]\370A\254\260\354\354"
  "A\330\331\254\260\354\20\0E#\217)a\260;\10\223\251\222\221\220\222\221\20R\61#$\42#$\242"
  "JD\225\214\204\224L\225\14\21\0F\33\217)a\223\213\222\213\222\233\21\34\221\264= ;\20\223\22"
  "\223\22\223\22\223\22G \221)g\70\60\221\70(\221\20\24Q'\242VdTFTfRHRH"
  "RJPJP\0H\16\15i]\370\240\36\313\3\241\3\21\0I\14\205)M\220\241\270\240\321o\0"
  "J\24\217)a\230\226^L{)q \63c$-.-\10K\35\221)\345\70\270\71\70\210\20\226"
  "\220\246\7\220\7\220\7\220\7\220\7\220\7\220&\226\0L\24\215)]\220\325\313\10\71\263\21\251\31\231"
  ")\221\261B\0M\20\213)]\70\272\234T*)\71)\71\251\0N#\216+\341\231\22\232\22\242\22"
[... 55 lines cut ...]
  ":=\15\22\15\222\24\222\224\222\233\232\251;\210\35\314\13\216)_\70\260\10\326\177\34\315\13\216)_"
  "\70\260\10\326\177\34\316\13\216)_\70\260\10\326\177\34\317\13\216)_\70\260\10\326\177\34\320\23\221)"
  "\347\70\70\10\71\70\240\346\325A\325A\65\327\2\321\33\214)]\70\240\70\240\20\322\225\214\224\214\324\210"
  "\230\210\230\210\234\204\234\204\240\0\322\13\216)_\70\260\10\326\177\34\323\13\216)_\70\260\10\326\177\34"
  "\324\13\216)_\70\260\10\326\177\34\325\13\216)_\70\260\10\326\177\34\326\13\216)_\70\260\10\326\177"
  "\34\327\13\216)_\70\260\10\326\177\34\330\14\216)\337(\351B\224\377T\0\331\13\216)_\70\260\10"
  "\326\177\34\332\13\216)_\70\260\10\326\177\34\333\13\216)_\70\260\10\326\177\34\334\13\216)_\70\260"
  "\10\326\177\34\335\13\216)_\70\260\10\326\177\34\336\24\214)a\20\325\323\203\240\203 )\31\61\21\71"
  "\11A\1\337$\223)k\20\71\70\30\71\270\20\21\225\20\221\234\207\220\207\230\207\220\207\230\207\220\7\21"
  "\22\21\223\22\21\223\2\340\13\216)_\70\260\10\326\177\34\341\13\216)_\70\260\10\326\177\34\342\13\216"
  ")_\70\260\10\326\177\34\343\13\216)_\70\260\10\326\177\34\344\13\216)_\70\260\10\326\177\34\345\27"
  "\221)\345\70\270\71\70\210\20\226\220f-v@v`\315\261\4\0\346\27\221)\345\70\270\71\70\210\20"
  "\226\220f-v@v`\315\261\4\0\347\35\221)i\70\250:\260\7\220\7\230\7\220\7\220\207\220\7"
  "\220\7\230\7\220\7\230\7\20\350\13\216)_\70\260\10\326\177\34\351\13\216)_\70\260\10\326\177\34\352"
  "\13\216)_\70\260\10\326\177\34\353\25\61Y\345:=\15\22\15\222\24\222\224\222\233\232\251;\210\35\354"
  "\13\216)_\70\260\10\326\177\34\355\13\216)_\70\260\10\326\177\34\356\13\216)_\70\260\10\326\177\34"
  "\357\13\216)_\70\260\10\326\177\34\360\24\221)\347\70\70\10\71\70\240\346\372\240\352\240\212\232k\1\361"
  "\32\214+\335\70\240\70 \322\215\224\214\224\310\224\210\230\210\230\204\234\204\34\241\0\362\13\216)_\70\260"
  "\10\326\177\34\363\13\216)_\70\260\10\326\177\34\364\13\216)_\70\260\10\326\177\34\365\13\216)_\70"
  "\260\10\326\177\34\366\13\216)_\70\260\10\326\177\34\367\13\216)_\70\260\10\326\177\34\370\14\216)\337"
  "(\351B\224\377T\0\371\13\216)_\70\260\10\326\177\34\372\13\216)_\70\260\10\326\177\34\373\13\216"
  ")_\70\260\10\326\177\34\374\13\216)_\70\260\10\326\177\34\375\13\216)_\70\260\10\326\177\34\376\21"
  "\214-\341\315WG\7\61b!r\21\202\204\1\377\13\216)_\70\260\10\326\177\34\0\0\0\4\377\377"
  "\0";
```

lcollins1989, 2020-09-01:
Hmmm. I’ll have a look at the BDF file i created and try again. Will report back!

Thank you so much

lcollins1989, 2020-09-01:
Ok, I've been trying again and apparently I'm completely incapable of using the cmd prompt.

I do however believe I have repaired the bdf file.

[Aurebesh.zip](https://github.com/olikraus/u8g2/files/5158194/Aurebesh.zip)

Is there any chance you could convert this one for me?

Thank you so much!

lcollins1989, 2020-09-01:
I’ve also played around and have no idea what to do with the finished file. Thought I’d try with the one that was corrupted and see if I could implement it in a code. I copied the c file to the u8g2 fonts file in the library but other than that I’m not sure what I need to do.

Sorry to be useless

olikraus, 2020-09-02:
looks better:

[image]

Here is the code:

``` C++
/*
  Fontname: Aurebesh
  Copyright: 
  Glyphs: 224/256
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_aurebesh[3676] U8G2_FONT_SECTION("u8g2_font_aurebesh") = 
  "\340\0\4\4\5\4\1\5\6\24\17\0\0\16\0\16\0\2\6\4\316\16\77 \5\0B\26!\22f\307"
  "\223\204\220\240d\4\211!$(\21$\10\0\42\17\351^\25\206\34\31\61\211\231B\244H\1#\21\316"
  "\303\227\14\332\350A{\320\17\32\31\264\21\0$\32\316\303WF\10\34!\354A\223\21B\210\214\20\62"
  "t(\321\231\22\235\16\0%\36\320\303\70\322%\33RlH\61'/\206\230\70Rf\304gd\206\25"
  "\31\226N\204\11\0&#\320\303\30\206\255:\61\310\314 \63d\14\15\71\64d\304\240!#F\235\30"
  "ed\224\221aE\206\61b'\10\345b\23\212\314\7(\14\305C\23\206\314\223&\363\4\0)\12\305"
  "CS\346I\223\371\0*\17\10[\65L\240\60\361 \204\11\24&\0+\16k\313\226\6\316\321\203C"
  "\3g\4\0,\7\303B\22\36\14-\7K\332\26\36\34.\11\307B\24F\330\213\1/\15\305CS"
  "\346\211\220)\206\314\23\0\60\24\317C\70\232\264\30i\67c\314\214\61i\313\21MZ\0\61\14\316\303"
  "\27\220!\226\377\354A\3\62\23\316\303\67\230\260\30hp\354l\37\204h\17\374A\3\63\27\316\303\67"
  "\230\260\30hp\354\330\26-\306\216\65hp\4\23\26\0\64\15\316\303\27\6\332\303\7\17\302\316\7\65"
  "\24\316\303\27\36\264\7\335\242\361\330i\15\32\34\301\204\5\0\66\22\316\303\27\36\264\7\316\204\305@{"
  "\70\202\11\13\0\67\12\316\303\27\36\264\235\377\3\70\23\317C\70Z<\30i\313\7\17L\332\362\301\210"
  "\26\0\71\21\316\303\67\230\260\30h\17G\60a\17\374A\3:\14\350\302\24H\221\32\66\305\203\0;"
  "\7\303C\22>`<\20K\313\66\205\225\71b\210\34As\347J\12=\13\313\322\26\36\234\7\367\340"
  "\0>\21K\313\26D\26\64g\220\34!\63f\212\15\4\77\16\306\303\23J<\30\63\177Bd\4\0"
  "@\35\316\303\67Z\260\30Ed\324\230\21\206F\224\32;E\251\21\206F\215\31E\206I\3A\34\321"
  "\303\31\210\216 \71\204\340\230\201\203Z\261\207\222Q\233\201C\10\216 yt\0B\30\321\303yV\265"
  "!G\202$y\350\226\255\207\216$\11rdZ\255\1C\14\312C\366\346\316D\336\315\35\0D\25\316"
  "\303\27>\20;x,\331d\310\310\16\36Kv\360\30\0E&\320\303\30\216\251\32ch\214\241\61f"
  "\6\35\31\64b\310\240\21S\215\230j\210\251!\246\206\24\33Cl\14!\0F\36\320\303\330\306\11\32"
  "'h\34\221\201$F\236e\246l\320\260A\303\6\15\33\364\340A\1G$\322C\32\134\214h\61\335"
  "\210!\303F\14\31:\204\344\230\221c\10\16\32\70h\340\250q\243\306\215j\306\0H\17\316\303\27\36"
  "\264\207\265\42\365\260~\320\0I\14\306\303\23\306\224x\63\377\6\0J\25\320\303\30H\217\246\270,\312"
  "\66D\16\215\36>z\364\62\5K!\322\303\71\336<\10\61v\304`\363\0\306\3\30\17`<\200\361"
  "\0\306\3\30lv\310\203 O\0L\30\316\303\27\306\316C\21\303P\21\31Df\10\251\21\304\14\226"
  "$\12\0M\23\316\303\27\220\241%;\361\330\261d\307\222\235\244\205\3N)\320\303\230\10\215\42\64\252"
  "\320\30Cc\14\215\71\63d\232!\323\14\31Bd\4\231!#\6\15\31\61\210\304)\23\303L\0O"
  "!\323C\272R.\34\65p\324\270\311\306\215\42Gh\344\240\221D\306\16\31;b\364\203\23\17D\0"
  "P\22\316\303\27F\225\30UbT\206\366\303\7!Z\0Q\20\320\303\70\234\270\30jt\364\374\35B"
  "\24\0R\25\316\303\27>\20;x,\331\301c\311\16\36Kv\360\30\0S'\320\303\30\6\222\30G"
  "d\334\230a\203FMDj\320\260\61\343\206\214\22\61\202\320\211\61D\214\224)Q\252\210\70\0T\26"
  "\320\303\330F\317WF\306L\61d\320\211a\11Q\232%\6\0U\23\320C\31\216\255\62\63\310\20\21"
  "\243\366\323\7DZ\0V\27\320\303\30\10\222 F\206\20\251\61\344\206\214\64Kx\364|\7\0W\22"
  "\322C:\36\4yP\330\376\361\210\7\42\36\204\0X\37\317C\270\212\226\64\71b\340\210q$\210\215"
  "\31E\206\320\250A\243\306\214\33\62n\304\203\7Y%\320\303\30\10\241\30\203d\314\10\62#H\14\32"
  "\61d\20\241a\203\206\21!\67d\340\10\222F\215\26&\6\0Z\37\322\303\31\306\3\30\17\200Y\253"
  "\201\203F\216\31:d<\200\361\0\306\3\30l\370\301\203\4[\13\305C\23\206\64\231\237\64\1\134\16"
  "\305C\23\206\314\233!c\304\14\231\3]\12\305CS\232\314O\232\14^\5\0\302\23_\5\0\302\27"
  "`\5\0B\22a\34\321\303\31\206\36\35\61r\310\300\61\215\330C\311\252\321\300\61#\207\14\35\61\224"
  "\0b\30\321\303yV\265!G\202$y\350\226\255\207\216$\11rdZ\255\1c\14\312C\26\306\315"
  "\213\134\230\233\7d\23\316\303\27\36\64%:)\241TH\211NJtB\0e'\320\303\30\206\255:"
  "\61\310\314 \63d\14\15\71\64d\304\240!#F\235\30ed\224\221aE\206\221\31Ff\10\0f"
  "\37\320\303\330F\217\22\66h\330\30bC\310!a\224n\314\270\61\343\306\214\33\63\356\301\3\1g$"
  "\322C\32F<\20\321b\62\22\223\15\31Id\344\230\221c\6\16\32\70h\34\241q\243\206\221b\306"
  "\14\0h\17\316\303\27\36\264\207\265\42\365\260~\320\0i\13\306\303s\206<\10\63\377\1j\27\320\303"
  "\230\11\17&K\324\340\231\66G\10\17&<z\224\62e\0k!\322\303\71\336<\10\61v\304\340\361"
  "\0\306\3\30\17`<\200\361\0\306\3\60lv\310\203 O\0l\26\316\303w\347\205@dCH\215"
  "!\64\212\310\60\22\3M\26%m\25\316\303\327P\245\32;t\354P\242C\211\216\35:\366\1\3n"
  "&\320\303\70\6\221\32DhP\241Af\6\231\31sf\12\42c\206LBd\66C&\32\61\202\320"
  "\20SC\214\15o!\323C\272R.\34\65p\324\270\311\306\215\42Gh\344\240\221D\306\16\31;b"
  "\364\203\23\17D\0p\17\316\303\67J\215(\225\225A\373\341\210\16q\17\320\303\70\234\270\30j\365\374"
  "\374\344A\0r\22\316\303\27\36\64%:)\321I\211NJtB\0s$\320\303\70\10\16\31\70f"
  "\334\240a\203H\315l\320\270\61#\4\21\71\64d\10\231\21c\212\230Z'\244\0t\26\320\303\330F"
  "\317WF\306L\61d\320\211a\11Q\232%\6\0u\23\320C\31\206\251Z\64\306\314 \243\366\323\7"
  "DZ\0v\27\320\303\30\10\222 F\206\20\251\61\344\206\214\64Kx\364|\7\0w\22\322C:\36"
[... 35 lines cut ...]
  "\63j\314\250AsDf\324\230Qc\206\15\31\66d\334\210q#\6\16\322\15\317C\30\134\210\245\377"
[... cut ...]
  "\377\355\203\20\0\366\15\317C\30\134\210\245\377\355\203\20\0\367\15\317C\30\134\210\245\377\355\203\20\0\370"
  "\22\317C\70,\61ab\244\375\313\21\226\230\60\1\0\371\15\317C\30\134\210\245\377\355\203\20\0\372\15"
  "\317C\30\134\210\245\377\355\203\20\0\373\15\317C\30\134\210\245\377\355\203\20\0\374\15\317C\30\134\210\245"
  "\377\355\203\20\0\375\15\317C\30\134\210\245\377\355\203\20\0\376\25\317\303\270\345;d\251F\11\32&f"
  "\234\20r\42\10\232\24\377\15\317C\30\134\210\245\377\355\203\20\0\0\0\0\4\377\377\0";
```

lcollins1989, 2020-09-02:
Amazing! Thank you so much!

As a quick question, is there an easy way to change the size of the font from the code or is it best for me to generate new BDF files and figure out the command prompt?

olikraus, 2020-09-02:
What exactly do you mean by "size"? If you need the glyphs with a different height, then yes, you need another bdf file.

lcollins1989, 2020-09-02:
Yes that is what I mean.

Thank you for your time and help, and for creating an amazing library!

lcollins1989, 2020-09-02:
Ok, I'm truly sorry to bother you again. I've spent another hour trying and I've finally worked out how to use the command prompt entries for bdfconv (it helped that I compared your command line with the help file that comes up haha)

However I now get the error "can not open bdf file" whatever I try.

Is there any chance you could convert this one final file for me?

I'd be forever greatful.

Thank you.

[aurebesh_small.zip](https://github.com/olikraus/u8g2/files/5163695/aurebesh_small.zip)

olikraus, 2020-09-03:
[image]

```
/*
  Fontname: Aurebesh
  Copyright: 
  Glyphs: 224/256
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_aurebesh_small[3053] U8G2_FONT_SECTION("u8g2_font_aurebesh_small") = 
  "\340\0\4\5\4\4\1\4\6\17\14\0\0\13\0\13\0\1\332\4J\13\320 \5\0\61\5!\17\224\263"
  "\234@\204\20IH\204\20I\60\0\42\20f\373\204`\6\21\42\220P\10\62\214@\0#\35\272q\235"
  "@\2\13$\260@\202!\4\21B\11A\204\230@\2\13$\260@\202\1\0$\23\272q\15\21Q!"
  "\22!\304P\70\21\6(\234P\0% \274\261\15\42\202:J\220\241\202(B\220%\4!D\10#"
  "\10)\202\240a\4:J\10\42\0&#\274\261\5\241\216*\42 !\202\21E\60\302\4\22P \202"
  "\4$\204 \1\11\21LP\303\4\225L\2'\10S\237\204!\304\1(\14\263\221\204@\302\311!a"
  "\2\0)\11\263\221\14q\201\204\70*\12f\373\14\42.!\2\0+\15\210\63\35\301D\3\215`\242"
  "\1\0,\6Qq\204\2-\6(;\5\10.\11T\261\204@\304I\0/\15\263\221\14\361\42\10!"
  "\2\11\23\0\60\24\273\221\215$\232\33RD\1\11\24\220\220\303\65\221\4\0\61\13\272q\5\203\14\24"
  "\77\242\0\62\14\272q\15$\230\13\62\204\24S\63\23\272q\15$\230\13\62HD\202\14r\270%\220"
  "\0\0\64\13\272q\205\0\305\260\312\360\0\65\20\272q\5J\21QP\310 \207[\2\11\0\66\20\272"
  "q\5\212\215Y\214\70\1\207[\2\11\0\67\10\272q\5*\303\77\70\20\273\221\215$\232\33r\270\14"
  "\211k\42\11\0\71\20\272q\15$\230\23p\270\301\204H\304`\12:\13e\321\4\201\4\12H\10\5"
  ";\6\261q\204\5<\16x\65\255a\210 H\60\342\10\33\0=\7X\67\5\10!>\17x\65\205"
  "\301\210#. \42\210\31\12\0\77\17\264\261\204!\212\20&<\21b\10!\0@\30\272q\15)\4"
  "\22$\20b\2\31(\310@\6\12\204\30\201\4I\1A\31\275\321\5!\205\20P\20\341\204Q(y"
  "@\222QD\70!\4$R\0B\30\275\321\235\243\222\21N\210`\203\7 \241\344\1\20p\220\204\220"
  "\1\0C\22\267\21\265\60\23$\20A\2\21$\20\301\302\14\0D\20\272q\5*\202\24\245AF\11"
  "\31\244H\0E(\274\261\205\242\216\12D\210\200\4\21B\30A\204\20&\30A\2\21F\20!\204\21"
  "D\210\200\204\31H\230\201\204\21\10\0F\30\274\261-\241\2\22J\30\301\4\21\256\300\244\204 J\30"
  "\241\204\321\0G\36\275\361\5E\202H$\10\301\2\11e \2\6\42`\60\302\5#\134@\202\5\224"
  "\24\2H\14\272q\5\352A\201\10\362\200QI\15\264\261\4A\206\10b\230\360\33\0J\24\274\261\5"
  "a\5\35eqB\34#\320\60\302\6\213\324\1K\17\275\321\215%\36\25\70\34\13\352\304\22\0L\25"
  "\272q\205 \303]\20\201\21\64H\60\303\4\42\24a\2\2M\23\272q\205\242\14\24R@!\5\24"
  "\62H!\222P\0N(\274\261\35\201\4\32F\240\201\4\21\42 A\204\20F\220@\204\21B\230`"
  "\204\20F\20!\2\22\204(\42\204\42\2\0O\31\276\361%\343\20\23*\260\300\4\22L\240\0\5\21"
  "P\20!\203\20\324\3P\21\272q\205\240\206\10\252(!\4\24\17\251\0\0Q\17\274\261\15%\226\20"
  "PP\361YqE\0R\17\272q\5*\202\24e\220\242\14R$\0S%\274\261\205\0\205\10N\220"
  "\300\204\11J\240\240\4\12H\250`\4\12\42\20\201\10\11f\220 \204\30h\220\300\0T\25\274\261-"
  "A\305\225\60\242\20D\64B\210\312\70\42\205\2\0U\22\274\261\205\302\216\22Hd\301\10*N\233P"
  "\2\0V\24\274\261\5\341\206\20, \201\2\23D\70\42\5\25\257\0W\23\276\361\15&\232\20\64\210"
  "`\307-\241A\64\302\4\0X\32\273\221\245\1\207\23B\60!\204\22F a\4\12*\30\241\4\21"
  "J\10\13Y\33\274\261\5\201\214\10\250\20\261\11D\10\201\304(\60A\4\23$@R\12\5\0Z\22"
  "\275\321\205\200\221J(\300`\2\34\42\340\260\335\0[\13\263\221\204@\16\11'\207\0\134\15\263\221\204"
  "@\302D\220@\204\20\7]\12\263\221\14$\304\5\22\2^\5\0\261\4_\5\0q\5`\5\0q"
  "\4a\31\275\321\205@\207\14$\300`\224I\36\220\204\224\11\60\230 \3\11T\0b\30\275\321\235\243"
  "\222\21N\210`\203\7 \241\344\1\20p\220\204\220\1\0c\20\267\21\205\300\302I \202\4\42H\310"
  "\302\1d\21\272q\5\12\205\23\60\230c\14\14P\70\221\1e'\274\261\5\241\216*\42 !\202\21"
  "E\60\302\4\22P \202\4$\204 \1\11\21LP\303\4\65LP\2\5\2\0f\31\274\261-A"
  "\5\12J\30\241\4\21\254\20\204\210\20* \241\2\22\12\3g\35\275\361\205@V\22\230\20A\12\21"
  "\240 \1\12\22`\60\301\11\23\234\60\11%\4\0h\14\272q\5\352A\201\10\362\200Qi\13\264\261"
  "\24!\212\10&\374\1j\26\274\261U!\205\34p\270B\216\10F(A\3\25\10\241\243\0k\17\275"
  "\321\215%\36\15\70\34\17\352\304\22\0l\25\272q\315\260\34.\10\301\2\31( a\202\32\42\270\1"
  "\5m\21\272q\255\202\14\22N@\341\4\24\60@\1'n'\274\261\15a\4\32f\30\201\206\11H"
  "\210`\2\22B\220\200\2\21$\30a\2\11F\24\202\4$DQ\201\20%\0o\31\276\361%\343\20"
  "\23*\260\300\4\22L\240\0\5\21P\20!\203\20\324\3p\20\272q\215\241\212\32\42(\1\305\341p"
  "K$q\20\274\261\15%\226\20p\330p+\250q\205\1r\17\272q\5\12\205\23\241p\42\24Nd"
  "\0s\37\274\261\15\341\4\21L\230\300\204\21J Q\11\63\224 \305\10\42\32!\204\31\204(A\6"
  "t\25\274\261-A\305\225\60\242\20D\64B\210\312\70\42\205\2\0u\22\274\261\205\300\214:H\64\201"
  "\11*N\233P\2\0v\24\274\261\5\341\206\20, \201\2\23D\70\42\5\25\257\0w\23\276\361\15"
  "&\232\20\64\210`\307-\241A\64\302\4\0x\32\273\221\245\1\207\23B\60!\204\22F a\4\12"
  "*\30\241\4\21J\10\13y\33\274\261\5\203\204(F\20\261\11D\10\201\304(\60A\4\23$@R"
  "\12\5\0z\24\275\321\345\240\20JF\270@\4\14\42\320\200\3\26v\3{\35\273\221\225\260\11\202\220"
  "\220\14\22D@\1\11\24\220@\1\15\62\10#\311\34U\14\0|\11\273\221\205+\305\227\27}\42\273"
  "\221\235\242\206\30\206\10\42\204\10&\210\242\214*\42\230 \216(\42\10Q\4\23\304\20A\25\3\0~"
  "\11\273\221\205+\305\227\27\177\11\203\223\5\42\304\13\2\200\5\0\21\4\201\5\0\21\4\202\5\0\21\4"
  "\203\5\0\21\4\204\5\0\21\4\205\5\0\21\4\206\5\0\21\4\207\5\0\21\4\210\5\0\21\4\211\5"
  "\0\21\4\212\5\0\21\4\213\5\0\21\4\214\5\0\21\4\215\5\0\21\4\216\5\0\21\4\217\5\0\21"
  "\4\220\5\0\21\4\221\5\0\21\4\222\5\0\21\4\223\5\0\21\4\224\5\0\21\4\225\5\0\21\4\226"
  "\5\0\21\4\227\5\0\21\4\230\5\0\21\4\231\5\0\21\4\232\5\0\21\4\233\5\0\21\4\234\5\0"
[... 12 lines cut ...]
[... cut ...]
  "\11\23\214@\301\10\24LP\201\10\25H`A\10\26Dp\303\5\362\11\273\221\205+\305\227\27\363\11"
  "\273\221\205+\305\227\27\364\11\273\221\205+\305\227\27\365\11\273\221\205+\305\227\27\366\11\273\221\205+\305"
  "\227\27\367\11\273\221\205+\305\227\27\370\22\273\221\15R\24q\334\220\342r\270#\212 \5\0\371\11\273"
  "\221\205+\305\227\27\372\11\273\221\205+\305\227\27\373\11\273\221\205+\305\227\27\374\11\273\221\205+\305\227"
  "\27\375\11\273\221\205+\305\227\27\376\22\273\221\325\360\312\240c\204\12D\260 \204\33\60\0\377\11\273\221"
  "\205+\305\227\27\0\0\0\4\377\377\0";
```

lcollins1989, 2020-09-03:
Thank you so much!

closed 2020-09-03 by lcollins1989